opentx/opentx #2468: Volume with GF can be set "dangerously" low

fraternl, 2015-07-14:
In each model I had a function to set the volume using the right-slider.
I made a curve for it, so it doesn't get muted.
This way I can always hear essential warnings like "A1 critical"

With "Global Functions" there's no need for it to be in the model.
I now directly connect it to the right-slider.
It is however too quiet when I have it totally down.

I still prefer it in this simple way, but it would be nice if it's a tad louder on its lowest volume.

I understand that all this is a matter of preference.

mpjf01, 2015-07-14:
There will be those who wish to be able to mute the warnings if they are subject to volume control, particularly during bench setup. As long as the volume control remains a matter for the user to explicitly decide to use it then he ought to be trusted to manage it himself.

fraternl, 2015-07-14:
Managing it myself means cluttering my model with things that are system related.
I am now in the process of moving as much as possible out of there...

I understand the bench setup (and the annoyed wife).
Another reason to prefer it system wide.

Maybe some kind of "nomute" checkbox?
We're talking about the prevention of a lost model here (RSSI low, battery critical)

mpjf01, 2015-07-15:
The way I see it there are two mutually exclusive options. One is to accept that these Tx generated alarms are critical and need to be heard regardless. This would exclude a "mute" capability and, arguably perhaps, should also exclude a volume control for these sounds.

The other option is to allow these alarms to be muted, either explicitly or by lowering the volume or both. I thought that you were arguing originally for this option with a limit to how low the volume could be set such that the alarms were always audible. 

Now, I think, you are suggesting that the alarms are too critical to mute, supported by others with the same feelings. If true, then they are too critical to limit the volume for as well as OpenTx knows nothing about the user's hearing ability/deficits or the level of interfering background noise. 

My point was that IF the decision is made to allow the alarm volume to be varied (I am not arguing for or against here) THEN the lowest level that it should be able to be set to should be user determined, not set in the firmware in stone for the reason stated above.

kilrah, 2015-07-15:
IMO the bench setup thing should be solved the way suggested in another issue, don't have the number at hand - a key combination on power up that disables all sounds.

Maybe the "volume" SF could be changed to adjust volume between "full" and the volume set in the radio settings. That way that setting becomes the de facto minimum.

fraternl, 2015-07-15:
Yes, that's a good idea....
The Special function is also not active during startup...

fraternl, 2015-07-23:
Would this work?

If volume set on radio is less than 50% than use it as the minimal volume output for the controller.
If it is set higher then the controller will be able to go mute...

A separate "minimal volume" in radio settings is still my preference.

closed 2016-11-27 by fraternl