opentx/opentx #2469: OpenTX 2.1 killEvents not working?

pinkywafer, 2015-07-15:
A simple test Telemetry script:

    local function run(event)
      lcd.clear()
      lcd.drawScreenTitle(" Test",1,1)
      killEvents(event)
    end
    
    return { run=run }

should display the screen title, and not respond to any key presses (as killEvents should catch and clear them), but all keys function normally.

projectkk2glider, 2015-07-18:
I think using `killEvents` is not meant to be used in the Telemetry scripts. What would you use it for?

pinkywafer, 2015-07-18:
The main use I see is multi-page telemetry scripts - using button presses to change pages.

There has also been a telemetry script which had several pages and options to send stick commands to cleanflight: http://rcsettings.com/index.php/viewdownload/13-lua-scripts/204-cleanflight-commands

projectkk2glider, 2015-07-18:
Most of the keys are already used by the radio only MENU is unused. @bsongis  and @kilrah  what do  you think, should we allow LUA scripts to take over the keys in Telemetry view? I could be dangerous.

kilrah, 2015-07-18:
For me all keys except EXIT and PAGE could be available to scripts. PAGE changes telemetry screens and we wouldn't want to block that, EXIT should always go back to the main views but the rest is unused in this context. ENTER LONG is used for the Reset menu, that could maybe stay for system use only as it might make sense also when using telemetry screens.

pinkywafer, 2015-07-18:
Thanks very discussing this guys.
At the moment, however + and - keys are switching telemetry screens as well as page and page long

pinkywafer, 2015-07-19:
Just a thought... I can see why you say to prevent EXIT and ENTER LONG being used in killEvents, but is the PAGE key important?  If not, you could then make a telemetry script with X pages, maintain the use of page to move through the screens until the last screen, then move to the next "native" telemetry page by not using killEvents at this point.  ie:

    if event == EVT_PAGE_BREAK then
        if currentPage +1 < numberOfPages then
            killEvents(event)
            currentPage = currentPage +1
        else
            --not needed as without killEvents next "native page is displayed"
        end
    elseif event == EVT_PAGE_LONG then
        if currentPage > 1 then
            killEvents(event)
            currentPage = currentPage -1
        else
            --not needed as without killEvents previous "native page is displayed"
        end
    end

kilrah, 2015-07-19:
It might be frustrating to the user as they might want to switch to the next telemetry screen NOW without having to scroll all the way through yours first.

So I'd rather keep PAGE for system use and use MENU for navigation between multiple pages of a given script. It also doesn't change what has kind of become the de facto standard for that in 2.0.

pinkywafer, 2015-07-19:
OK, that makes sense.
if it's possible to use the + and - and ENTER (not long) too that would be nice!

kilrah, 2015-07-19:
Yes, as I said to me everything but PAGE and EXIT (short and long) and ENTER long could be available to scripts.

bsongis, 2015-07-19:
I would see only + - and ENTER (not long) available. It should work this way, no?

dsbeach, 2015-11-10:
I'm digging into this today - it's something I'd really like to solve for my own telemetry scripts that process keys (+ - ENT and EXIT).

It appears to be related to the deprecation of lcd.lock() in 2.1.

Here is a relevant section of 2.0 code from opentx.cpp in perMain(). If a Lua script locks the lcd the key event is not passed on to the menu function that advances the telemetry screen
```c
    if (!LCD_LOCKED()) {
      lcd_clear();
      g_menuStack[g_menuStackPtr]((warn || menu) ? 0 : evt);
    }
```

Under 2.1 in main_arm.cpp function perMain() the menu function is invoked regardless of the key state (around line 185)
```c
    g_menuStack[g_menuStackPtr]((warn || menu) ? 0 : evt);
```

So... What is the best approach to a fix? What was the logic behind deprecating lcd.lock()? Should it be implemented now, or should there be code added to correctly process the killed() events?

dsbeach, 2015-11-10:
Testing the above pull request looks good so far.

Below is a Lua script to try.  It appears that this fix enables all keys to be processed in a script except for:
  -  EXIT - not seen by the script and immediately exits telemetry screen display mode
  -  repeat keys (EVT_PLUS_RPT and EVT_MINUS_RPT) are not seen

I like the idea of allowing Lua scripts to process the page key events. I think it's OK for a script developer to create a multi pages telemetry script as long as it behaves in a predictable manner.

```lua
local lastKey = 'none'
  return {
    run = function(event)
      if event == EVT_EXIT_BREAK then
        lastKey = "EVT_EXIT_BREAK"
        killEvents(event)
      elseif event == EVT_MENU_BREAK then
        lastKey = "EVT_MENU_BREAK"
        killEvents(event)
      elseif event == EVT_PAGE_BREAK then
        lastKey = "EVT_PAGE_BREAK"
        -- killEvents(event)
      elseif event == EVT_PAGE_LONG then
        lastKey = "EVT_PAGE_LONG"
        -- killEvents(event)
      elseif event == EVT_ENTER_BREAK then
        lastKey = "EVT_ENTER_BREAK"
        killEvents(event)
      elseif event == EVT_ENTER_LONG then
        lastKey = "EVT_ENTER_LONG"
        killEvents(event)
      elseif event == EVT_PLUS_BREAK then
        lastKey = "EVT_PLUS_BREAK"
        killEvents(event)
      elseif event == EVT_MINUS_BREAK then
        lastKey = "EVT_MINUS_BREAK"
        killEvents(event)
      elseif event == EVT_PLUS_RPT then
        lastKey = "EVT_PLUS_RPT"
        killEvents(event)
      elseif event == EVT_MINUS_RPT then
        lastKey = "EVT_MINUS_RPT"
        killEvents(event)
      elseif event == EVT_PLUS_FIRST then
        lastKey = "EVT_PLUS_FIRST"
        killEvents(event)
      elseif event == EVT_MINUS_FIRST then
        lastKey = "EVT_MINUS_FIRST"
        killEvents(event)
      end
      lcd.clear()
      lcd.drawText(10, 20, "idx : ", 0)
      lcd.drawText(lcd.getLastPos(), 20, lastKey, 0)
    end
  }```

projectkk2glider, 2015-11-25:
Reopening until I make cosmetic changes and port this to the next branch

projectkk2glider, 2015-12-13:
Reopening till the Horus main gets fixed. The commit https://github.com/opentx/opentx/commit/806bd1da53896ace6066505c9043d6b0108de159 did not really solve this, in fact it is not even in the main branch. Github got confused somehow.

dsbeach, 2015-12-13:
Wasn't this fixed by #3112 ? Main is working, and I'll test next later today.

projectkk2glider, 2015-12-13:
>Wasn't this fixed by #3112 ? Main is working, and I'll test next later today.

It was for Taranis, the Horus uses different GUI logic and I haven’t managed to port the changes yet.

projectkk2glider, 2016-10-03:
Closing, Horus solution in @3821.

commit 3186a5d9 (2015-11-21): Fixes #2469: Allow usage of PLUS, MINUS, MENU and ENT keys in telemetry scripts
PR #3087 (2015-11-24): Fixes #2469: Allow usage of PLUS, MINUS, MENU and ENT keys in telemet…
…ry scripts

I think this is the right solution:
* all keys are seen by the Lua scripts
* Lua scripts can not use killEvents() on ENT and EXIT
* if a telemetry script has run, filter out the keys PLUS, MINUS, ENT (only short),  EXIT (only short) and MENU from the menu system. These keys can now be used freely in the telemetry scripts.

Tested in simu.

Closes #2469. Replaces #3051

Updated description (added EXIT short)
commit d464420a (2015-11-24): Merge pull request #3087 from opentx/projectkk2glider/issue_2469_lua_keys

Fixes #2469: Allow usage of PLUS, MINUS, MENU and ENT keys in telemet…
commit 806bd1da (2015-12-12): Added logic in perMain() to suppress killed events.  Fixes #2469
closed 2016-10-03 by projectkk2glider