opentx/opentx #2470: Feature Request: Grouped global switches

fraternl, 2015-07-16:
I really like the new "Global Functions" and have already taken stuff out of my model and put it there.
It does emphasize my need for a way to do the same for switches.

In https://github.com/opentx/opentx/issues/2097 I suggested a radio-wide calibration system. Due to the open nature of OpenTX it's probably hard to implement in an elegant way.
When I implement Mike Shellim's calibration method it's not the mixes themselves that I feel are cluttering the model. It's the set of logical switches using "sticky".. (sticks in special position and 

If it were possible to have predefined "Global switches" (1 to 8?) that each consist of a group of switches like you can have in a model one would have an organized set of switches....

The instant-trim, for instance, could be triggered by such a switch. Doing this with the momentary switch SH is quite dangerous.

By having it organized in your radio you can have the same method for triggering stuff.

kilrah, 2015-07-16:
Duplicate of #2409.

closed 2015-07-16 by kilrah

fraternl, 2015-07-16:
That was in fact a duplicate of https://github.com/opentx/opentx/issues/2121
Doesn't matter.
Looking forward to it.