opentx/opentx #2472: Global Function BGMusic can't select a track

fraternl, 2015-07-16:
The Global Function BGMusic can't select a track

kilrah, 2015-07-16:
In companion? On the radio?

Please work a bit on making usable issue reports.

fraternl, 2015-07-16:
Sorry....
On the radio.

bsongis, 2015-07-16:
No problem for me, someone else?

fraternl, 2015-07-16:
Bernard,
Can you select a track?

http://hmvc.eu/2015-07-16 19.58.26.mp4

bsongis, 2015-07-16:
Bertrand :)

fraternl, 2015-07-16:
Sorry...   My French butcher is called Bernard  ;-)

bsongis, 2015-07-16:
Yes I can select a track without problem!

fraternl, 2015-07-16:
Just did a factory reset and it's still the same.
Can you see me do something wrong in the video?

bsongis, 2015-07-16:
Which is your language selected?
Or perhaps because your filename begins with a '3'?

bsongis, 2015-07-16:
Would you put somewhere your eeprom + a zip with your SD contents?

fraternl, 2015-07-16:
I have yet another problem...
My SF3 changes from time to time to the first track (3dhobbyshop).
I had this before, but thought it was some user error.
Now it changed again.

fraternl, 2015-07-16:
The firmware and eeprom
http://hmvc.eu/EEPROMS.zip

fraternl, 2015-07-16:
I thought it might have been the large music files I have in /SOUNDS/en/, but I just formatted the SD-Card in my PC and copied a backup without those music files to it.
I checked Global Functions again and it did the same.

I haven't tested enough to see if SF3 has changed again.

http://hmvc.eu/Taranis_SD.zip

fraternl, 2015-07-16:
I now tested it and when I select a track in GF3 it will be written to SF3

bsongis, 2015-07-16:
Sorry I didn't see you were talking about Global functions. Now I reproduced the problem with your firmware.

fraternl, 2015-07-16:
;-)
I'm so glad these 2 are related.
I was afraid of some memory overwrite like with the LS1

closed 2015-07-16 by bsongis

bsongis, 2015-07-16:
It was just a stupid bug, something that obviously nobody tested. I knew OpenTX 2.1.1 would be quickly needed ...

fraternl, 2015-07-16:
I started using Global Functions 2 days ago. 
BTW...   None of the translations are implemented. That's intended or not?

commit 92b5b723 (2015-07-16): Fixes #2472
commit b05e77a2 (2015-07-16): Fixes #2472