opentx/opentx #2475: Trimmer sound intereface bug

jetigm, 2015-07-16:
Push the trim button, on screen it goes to the end with according sound, then when it in the end the sound is ends but if to continue to push the trim button the trim sound begins again and sounds while the button is pressed.
I think this in don't logical, I want to hear the trim sound while the trim button is pressed but when the trimmer comes to the end I think the sound should stops and don't repeat again even if the trim button still pressed for a while.

kilrah, 2015-07-16:
The trim keys can be used for many other things than trimming, in which case the sound is useful. You do get the "maximum trim reached" to cover the standard case.

fraternl, 2015-07-19:
If you're working on trim could you look at https://github.com/opentx/opentx/issues/1814 and https://github.com/opentx/opentx/issues/1813  

With a SF one can disable the trim by letting it change a Global variable.
A sound is not always wanted in such a case (especially not the trim centre and trim maximum)

closed 2020-07-08 by stale[bot]