opentx/opentx #2476: Companion 2.1 not saving vario source and centre silent options

helicop, 2015-07-16:
Opentx-taranis-plus-2.1.0, 17/07/2015 EEPR:217

Reading taranis memory to companion, then writing back, variometer source is blank (-----) and the centre silent option has changed to "Tone". Very minor problem, just have to remember to change things at next use.

bsongis, 2015-07-16:
Would you check for me something? I think there is also a problem with volts and current sources

helicop, 2015-07-16:
The only sensor I have with voltage and current is a Unisens-E. I have tested it, and the voltage and current are working fine

kilrah, 2015-07-16:
What he means is the selection for "voltage source" and "current source" on the telemetry page. Regardless of the sensors you have.

bsongis, 2015-07-16:
*posts removed, I read the code too quickly

commit 158aea98 (2015-07-16): Fixes #2476
commit 475f1f1b (2015-07-16): Fixes #2476
closed 2015-07-16 by bsongis