opentx/opentx #2477: Not really a bug, but timestamp of downloaded firmware never changes unless I change the name

fraternl, 2015-07-17:
Because I want to have the latest fix of the radio-firmware as a file on the SD-Card I did a "download firmware" with today's companion.
It took some time before downloading indicating a compilation on the fly (is that correct, I'm assuming that).

I went to the folder where I keep those firmwares and did a sort of the files by date...
The file I just downloaded had a timestamp of 29-6-2015.
Yesterday's file (one I renamed in companion before downloading) had yesterday's timestamp.

I then deleted the file with the default name (opentx-taranisplus-lua-mixersmon-en-2.1.0.bin) and the old timestamp. I did another download without changing the name.
Again the file was created and again with 29-06-2015 as a timestamp.

I then did yet another download and renamed the file before downloading to opentx-taranisplus-lua-mixersmon-en-2.1.1.bin
This time It got today's timestamp.

Is all this intended behaviour?
I think so, as you need to do something extra to change the filedate.

kilrah, 2015-07-17:
Can't reproduce, all the files I've downloaded these last few days (always with default names) have dates from when I've downloaded them. 
Must be something on your system as it has nothing to do with us. Companion creates a file, the system gives it the attributes it wants.

closed 2015-07-17 by kilrah

fraternl, 2015-07-17:
Can't be something with my system. I continuously check dates of files.
But not important really.

bsongis, 2015-07-17:
Which one did you download?

fraternl, 2015-07-17:
I downloaded 17-07-2015 here this morning. 
But it's the firmware I download with it, not the download of Companion.

It will get the name "opentx-taranisplus-lua-mixersmon-en-2.1.0.bin" and asks me to overwrite. Which I do. It will download it and give it a timestamp of  29-06-2015
That's when I first downloaded the firmware on this machine.
I've been doing that each day since.

I now got rid of this behaviour by removing it again and downloading it.
It now has the timestamp 17-07-2015 08:16
If I download it again and tell it to overwrite, the timestamp stays the same
I just removed the file downloaded it and again it's 17-07-2015 08:16

Somehow it must have changed to today, but it's stuck again....

kilrah, 2015-07-17:
That's perfectly normal system behavior.
If you overwrite the file it is still the same file form the system's point of view, so its creation date is still the one of the first time it was created. Only its *modification date* will be set to now.

fraternl, 2015-07-17:
You probably missed the part where I deleted the file.
On Windows the last modified timestamp is shown

fraternl, 2015-07-17:
I guess you're right after all
I can choose out of 3 column headers:

"Datum", "Gewijzigd" and "Aanmaakdatum"  (Date,  Modified, Created)

I only had the column "Datum" selected
It's only in THAT folder.
I have no clue why that column is chosen and why that ambiguous name is there (Windows 8.1)
I thought Windows only used 2 timestamps

fraternl, 2015-07-17:
[image]

fraternl, 2015-07-17:
Just Googled a bit.... This is new to me.

"The "Date" column takes the earlier of "Date created" and "Date modified", ignoring "Date accessed","

Not too old to learn a new trick.
You knew?

http://superuser.com/questions/147525/what-is-the-date-column-in-windows-7-explorer-it-matches-no-date-column-from