opentx/opentx #2478: Something happened to SF3 after changing GF3 (after yesterda's fix)

fraternl, 2015-07-17:
Today I did a new test containing yesterday's fix

SF3 is triggered by FM1 and says "launch mode"

If I change the BGMusic on GF3, the GF3 worka as intended, but it doesn't say "launch mode" anymore.

When I check the SF3 it still seems correct. It still displays "launchmd", but it just doesn't play,
Even if I change it to something else.

Only when I clear GF3, the SF3 plays

I renewed the test....
I re-entered a GF3 with BGMusic and the SF3 doesn't play. Even if I didn't select a track in GF3 yet.

fraternl, 2015-07-17:
Oh... I tried a "Play Track" in GF3 and then it didn't occur...

bsongis, 2015-07-17:
Again I am not sure I understand the issue.

fraternl, 2015-07-17:
SF3 is set to play "launch mode"
If I put a BGMusic on GF3, the SF3 stops working .

Nothing strange can be seen on the display, but it just stops functioning

kilrah, 2015-07-17:
What are you triggering GF3 with? Same switch as SF3 or a different one?

fraternl, 2015-07-17:
Different One

fraternl, 2015-07-17:
Can you already confirm the issue?

pinkywafer, 2015-07-17:
I've just had a similar issue:

ONLY occurs on the TX (and simulator) NOT when set on companion

ANY track selected for a "PLAY Track" OR "BGMUSIC" is NOT stored in the relevant variable for a global function, but is instead stored in the model function:
Global Function 1 saves the track variable to Model Special Function 1
Global Function 2 saves the track variable to Model Special Function 2
etc

To reproduce:
on simulator or TX
1. create a "play track" or "bgmusic" MODEL Special function at SFx and select the track
2. create a "play track" or "bgmusic" Global function at GFx and select a track different from above

The track selected at 2 will be saved at 1 instead of 2

If set up on the companion they work fine.

pinkywafer, 2015-07-17:
Apologies, the above is SIMULATOR ONLY, it works fine on the TX.  I'll raise a seperate issue, BUT:

if bgmusic is playing, any track should play "over" the bgmusic....
If the bgmusic is on the same global function number (ie GF1) as the play track (SF1), the SF track is NOT played over the bgmusic

fraternl, 2015-07-17:
The BGMusic in GF3 doesn't need to play to disable play track in SF3.

In GF3 I have a BGMusic on a switch that's disabled and never played. SF3 week then not play a track.

I never used the simulator

bsongis, 2015-07-18:
Ok, I reproduced something similar

closed 2015-07-18 by bsongis

fraternl, 2015-07-20:
The original issue is not fixed.

Bertrand,
I seem to have difficulty reporting to you.
What would be the best way to report it?
Should I make a video?

kilrah, 2015-07-20:
Where is it not fixed? What build from what day and what source? We've got releases from companion, nightlies, self-compiles from different potential branches...

It's not only to him... and the best way is to ALWAYS include all information about what versions you're using and what exact steps you're doing to cause the problem so we can reproduce it. Otherwise we're taking a wild guess and can only hope we catch your particular case while fixing what seems wrong.

projectkk2glider, 2015-07-20:
@fraternl stop using 2.1.99 from `next` branch. See https://github.com/opentx/opentx/issues/2521#issuecomment-122982041

commit 2b804f92 (2015-07-18): Fixes #2478
commit 1da6aee9 (2015-07-18): Fixes #2478