opentx/opentx #2480: Maintenance mode is not working in 2.1 released

fred99, 2015-07-17:
May be I do something wrong but I have updated the firmware of my Taranis to 2.1:
vers : opentx-taranis-2.1.0
date : 2015-07-17
And I am unable to enter in maintenance mode" :
I hold both horizontal trims AWAY from the centre while powering on without success.

kilrah, 2015-07-17:
There is no "maintenance mode" on OpenTX and there have never been any plans to have one. 
All the operations you would likely expect to be there coming from other firmware are done from the SD card browser.

closed 2015-07-17 by kilrah

fred99, 2015-07-17:
May be my wording is not good in release notes we have:
[Taranis]
Added firmware upgrade of SPORT devices (#1599)

Is it not a kind of maintenance mode?
The goal is to update TX and RX using s.port.
How we can use this new possibilitie?

kilrah, 2015-07-17:
As I said, go to the SD card browser in radio settings, navigate to your frk file, press enter, and select flash s.port device (internal or external).

fred99, 2015-07-17:
Ok, it's clever and better than maintenance mode! 
Thanks a lot for your help, I will try it and share this information.