opentx/opentx #2481: Firmware simulator 2.1 Global Function "track" variable saved in wrong place

pinkywafer, 2015-07-17:
ANY track selected for a "PLAY Track" OR "BGMUSIC" is NOT stored in the relevant variable for a global function, but is instead stored in the model function:
Global Function 1 saves the track variable to Model Special Function 1
Global Function 2 saves the track variable to Model Special Function 2
etc

To reproduce:
on simulator or TX
1. create a "play track" or "bgmusic" MODEL Special function at SFx and select the track
2. create a "play track" or "bgmusic" Global function at GFx and select a track different from above

The track selected at 2 will be saved at 1 instead of 2

If set up on the companion or TX they work fine.  This only occurs in "simulate TX" or on the standalone simulator

fraternl, 2015-07-17:
The behaviour you are describing is like the radio was doing yesterday.I reported this and Bernard fixed it.
But as it now seems only partly.

Maybe the simulator you are using is not yet up to par

bsongis, 2015-07-18:
Duplicate of #2478

closed 2015-07-18 by bsongis