opentx/opentx #2485: Can't register individual cells correct in Companion 2.1

ReneTr, 2015-07-17:
I have a Taranis X9D plus, an X6R with a FLVSS.

The FLVSS voltage sensor registers fine and cels shows correct values.

When in Companion 2.1 I add a new sensor: lowc Calculated Cell Cells Lowest
I want to use that sensor in a Logical Switch for a lowest cell warning: Like a<x lowc 3.50v
But I can only select integers in the V2 column, it doesn't know it's volts. In the telemetry screen it also shows an integer. Example: 4 instead of 4.12

If I create the lowc sensor on the Taranis it works fine. Also if I then read the settings from the Taranis into Companion, Companion works fine as well. It now knows it's Volts and I can create the Logical switches I want.

ReneTr, 2015-07-17:
[image]

kilrah, 2015-07-18:
Was related to #2479

commit d11f38f0 (2015-07-18): Fix #2485
commit cad3b772 (2015-07-18): Fix #2485
closed 2015-07-18 by kilrah