opentx/opentx #2486: Companion v2.1.0: Telemetry screens not populated after model loaded

Bamfax, 2015-07-17:
Hi, 

when I reload my Models and Settings from file or from the receiver, the Telemetry screens (two configured, as numbers) do not populate the fields in the Model Editor / Telemetry Screen Edits. 

Nevertheless, the setttings are all loaded, are configurable and are saved just fine. The configured telemetry fields just do not show when the model was loaded.

Thanks again,
Olli

bsongis, 2015-07-18:
Do you have the official Companion 2.1.0? Configured for which board?

kilrah, 2015-07-18:
Should be all good... please elaborate on what versions you have and what exactly you're doing.

[image]

Bamfax, 2015-07-18:
It's a taranis (maybe 2nd edition) running OpenTX 2.1.0 and Companion 2.1.0.

Maybe something got garbled when the model was upgraded or me playing around with the telemetry values. I can send you the settings file if wanted.

The fields which are configured on the Companion telemetry screen do not stay populated when closing the model and reopening it, or when closing the models/settings file completely and reloading it from disk or the receiver. The the tele screens keep their type (numbers/bars), but the fields are always empty in the editor, regardless of what is really configured. The simulator also does not display the configured fields. This is the case on all models.

[image]

[image]

kilrah, 2015-07-18:
Really can't reproduce. 
So yes please upload your settings file somewhere and give us a list of actions to do in companion with it that would recreate the problem.

Bamfax, 2015-07-18:
https://github.com/Bamfax/Testing/blob/master/Taranis%20Konfig%202.1.0b%202015-07-17j.eepe

Basically just open the Model "Tester", that is the one shown above on the pictures, goto the tele screens, they then should be empty as in the picture.

bsongis, 2015-07-18:
Would you send a bin from the radio instead of an eepe from Companion? Also what is the date in the About page of Companion? I can't reproduce this bug.

projectkk2glider, 2015-07-18:
I just reproduced on 4ad6ea8190e06267e17b69f8178f60ff7fa5e9f7, will try lastest master now..

kilrah, 2015-07-18:
They are empty, but if I add something it stays there. Please explain what you're doing for them to get lost.

projectkk2glider, 2015-07-18:
I think he has them configured on the radio, maybe they are not imported into Companion.

bsongis, 2015-07-18:
He sent an eepe, we really need the bin from the radio, not something exported by Companion. My guess is he uses an old Companion (former to 17 of july) which had this bug.

projectkk2glider, 2015-07-18:
Aha, that makes sense, so his eepe is already empty there.

Bamfax, 2015-07-18:
Bsongis is right, I am using Companion 2.1.0, 1st of July. Sorry, I just did an update check which came with no updates available.

Here is the bin:
https://github.com/Bamfax/Testing/blob/master/Taranis%20Backup%202.1.0b%202015-07-17j.bin

projectkk2glider, 2015-07-18:
It is all good with latest (not yet released) Companion. Just tested above .bin.

closed 2015-07-18 by projectkk2glider

kilrah, 2015-07-18:
> Bsongis is right, I am using Companion 2.1.0, 1st of July. Sorry, I just did an update check which came back with no updates available.

That means you're using an outdated prerelease build. 2.1 has since officially been released so you should go and download the final version from the open-tx.org website. Prerelease builds do not give update notifications.

Bamfax, 2015-07-18:
Well sorry guys for not noticing that beforehand. And thank you for the help.