opentx/opentx #2487: OpenTX 2.1 Telemetry: Date not logged correctly

Bamfax, 2015-07-17:
Hi,

the date field has the same issue as the GPS field when being logged (as in issue #2456 ). Its fieldname is logged as "Date(d<)" and the value stays at 0, even though GPS time is received via telemetry and displayed correctly.

Thanks again for fixing,
Olli

Bamfax, 2015-07-17:
Cels and calculated CelL do not exhibit this behavior, they work fine.

Bamfax, 2015-07-17:
The closed #2456 caused a little sideeffect, maybe you can revisit that. See comment in #2456. Let me know if you want a separate issue for that.

nscch, 2015-07-18:
Hello
The voicegenerator didnt response the right units the telemetry data. E.g. the alt (meter) response in "miliampere per hour"

kilrah, 2015-07-18:
Unrelated, make sure you've updated your voice files.

nscch, 2015-07-18:
How can I get the weakest cell voltages from FLVSS in the data. I get only the main voltage from the Lipo.

Thank you very much for your answer.

commit cd3d473d (2015-07-18): Fixes #2487
commit e1e60b55 (2015-07-18): Fixes #2487
closed 2015-07-18 by bsongis