opentx/opentx #2491: Bars ranges not transfered to radio

projectkk2glider, 2015-07-18:
When I defined bars in Companion like this, I get empty bars screen in simulator (both ranges are 0-0). 
[image]

kilrah, 2015-07-18:
Just read about someone wishing that we allow for having start > end and have the bar work the opposite way e.g. to use mAh count to show "remaining" instead of "elapsed", maybe we can take care of that at the same time.

bsongis, 2015-07-18:
Good idea, but he can open another issue, I prefer doing only fixes right now to have a good start for 2.1 ...

closed 2015-07-18 by bsongis

projectkk2glider, 2015-07-18:
>Good idea, but he can open another issue, I prefer doing only fixes right now to have a good start for 2.1 

Done in #2495

commit 34731af0 (2015-07-18): Fixes #2491
commit 7be388f7 (2015-07-18): Fixes #2491