opentx/opentx #2493: update to 2.1 leaves phantom sound file links

0gopogo, 2015-07-18:
.downloaded and installed 2.1
.gearup and geardown speech were assigned to SFup and SFdown in 2.0
.gearup remains
.geardown is erased
.no reference to this sound file appears in companion
.no assignment for switch SF remains in Tx in Special Functions

kilrah, 2015-07-18:
Can you upload your 2.0 backup .BIN file?

0gopogo, 2015-07-18:
I hope this is the correct file. I note that I have a sound directory under the airplane name that contains the switch position and file to be played - rather than special functions. SF-up plays but SF-down will not. I am about to replace the file in case it is corrupt.
 
cheers
john 
 
Date: Sat, 18 Jul 2015 11:15:21 -0700

0gopogo, 2015-07-18:
renamed SF-down to SF-mid and it works :) so 2 pos switch is addressed as Up and MID

0gopogo, 2015-07-18:
sorry for the mis-direction on Special Function - I know your time is valuable and greatly appreciate your work here.

bsongis, 2015-07-18:
It's "only" an EEPROM conversion error in the firmware, not in Companion, right?

0gopogo, 2015-07-18:
I would think so as the sound file is read directly from the SD card. Unless there is something happening behind the scenes in companion to link the switch action and sound file.

0gopogo, 2015-07-18:
In 2.0 switch SF was read as Up and Down. In 2.1 it appears to be Up and MID.

bsongis, 2015-07-18:
I just understand now, some users are really difficult to understand. Unless it's me!

closed 2015-07-18 by bsongis

0gopogo, 2015-07-18:
ok, so who would address this change ?

kilrah, 2015-07-18:
?

Should have been fixed by the commits referenced above. You can try the upcoming nightly build to confirm.

0gopogo, 2015-07-18:
ok, I understand the comments now - thanks
john

commit 18f28270 (2015-07-18): Fixes #2493
commit 3e397606 (2015-07-18): Fixes #2493