opentx/opentx #2498: Feature Request: PPM input on serial port

pinkywafer, 2015-07-18:
Could not find this raised before... would it be possible to have PPM input on serial port for trainer? This would allow me to use a D4R-II in the battery compartment with just a single connection, and powered from the TX

bsongis, 2015-07-19:
It is already possible to have SBUS input there, it would do the job! For the moment no PPM though, but it would be possible.

pinkywafer, 2015-07-19:
It would be nice... I would like to use the D4 in the battery compartment instead of the X4, in trainer configuration, as a some people use the old frsky module (dft) which doesn't support the X series, and it would be nice in the battery bay, to still allow the use of the external module, and not needing extra wiring etc outside the case (ie trainer port)

LapinFou, 2015-11-10:
The new Sbus/CPPM trainer modes are fantastic. Being able to do some wireless trainer is nice.
I discovered this tiny 8 channels receiver: MicroFrX
With some small extra electronic parts (to adapt the power supply), it will perfectly fit in the battery compartment. However, the COM port currently supports only the SBus protocol. Would it be possible to add the CPPM mode?

Thanks,
Seb

closed 2020-07-08 by stale[bot]