opentx/opentx #2500: Simulators missing from Companion installer on Linux platforms

projectkk2glider, 2015-07-19:
Two problems:
 * The simulator dynamic libraries libpentx-*-simulator.so are missing from RPM and DEB packages. 
 * The Companion looks in current working directory for those files, which is not the standard way on Linux, the libs should be in /usr/local/lib/companion.

Perhaps similar problem is also present on MAC.

This issue is solved in pull #2482, which also includes a change that enables Companion 2.0 and 2.1 install at the same time (both packages can't have the same files).

@bsongis and @kilrah I urge you to consider that pull request before the 2.1.1 is released!

bsongis, 2015-07-19:
I think this one can be closed, right?

bsongis, 2015-07-21:
@projectkk2glider wouldn't it be better to use companion21 name instead of companion2.1 ? A little bit like python ...

projectkk2glider, 2015-07-21:
Sure, the dot was bothering me also. I will change it...

bsongis, 2015-07-21:
Gah, I will continue working on Linux then :)

projectkk2glider, 2015-07-21:
@bsongis Please test my work and close the issue.

bsongis, 2015-07-21:
Yes, it's perfect now, thanks!

PR #2482 (2015-07-19): Enables side by side installation of Companion 2.1 and 2.0 on Linux
Adds missing simulator libraries on Linux
Installs Standalone Simulator icon on Linux

I've only tested this on Linux, I might have broken something on other platforms.

The simulator library search path is hard coded, which is not ideal solution. But I have no better idea. Without this change the simulation does not work and in fact the official RPM package for 2.1 is broken.

This also enables one to have installed at the same time both the old Companion 2.0 and the new one 2.1 which I think is important.

I am not a cmake e [... cut]
closed 2015-07-21 by bsongis