opentx/opentx #2502: Companion 2.1 doesn't show calibration settings for pot 4

franck76, 2015-07-19:
my radio : Taranis with Opentx 2.1 official release

Start Companion from nightly-21  18-jul-2015
Read models ans settings from my radio

Calibration settings for pot4 are missing, they are set to 0 for other sticks and pots :
[image]

do the same with companion 2.0, yo get this : 
[image]

this may be a bug ?

bsongis, 2015-07-19:
Someone confirms?

kilrah, 2015-07-19:
What radio type?

franck76, 2015-07-19:
Type ?
Taranis 1st version (june 2013)

franck76, 2015-07-19:
I see, in both companions radio type is Frsky Taranis
My EEPROM was converted from the radio when installing companion 2.1 
I noticed that some models (not all of them) were not properly converted
I shall install a new eeprom and carry out calibration if nobody can reproduce

projectkk2glider, 2015-07-19:
Maybe @franck76 converted EEPROM in Companion. Because if I convert my 216 eeprom in Companion I also loose calibration settings (all become zero). Should I create a new ticket for this?

 The pot4 is disabled for Taranis and Taranis+, it is only enabled for X9E, which I think is ok.

franck76, 2015-07-19:
In the first place I am pretty shure I converted my EEPROM from the radio 
Then I also thried and converted the same EEPROM from companion
Anyway, I can do some further tests tonight

By the way I thought that Pot1-4 were S1, S2, LS and RS ?

kilrah, 2015-07-19:
I can confirm that calibration settings aren't displayed in either case coming from a 216 or 217 eeprom.

[image]

bsongis, 2015-07-19:
Someone would check, just one line forgotten?

closed 2015-07-19 by bsongis

franck76, 2015-07-20:
Seems OK with nightly from today :
[image]

commit 7336d92a (2015-07-19): Fixes #2502
commit f2bf35c1 (2015-07-19): Fixes #2502