opentx/opentx #2503: Standalone simulator radio selection display problem

projectkk2glider, 2015-07-19:
Two problems:
* window width on narrow displays 
* buttons to small to display all text

[image]

bsongis, 2015-07-19:
Yes a combo box would be really better, would you take care of it?

projectkk2glider, 2015-07-19:
Ok

kilrah, 2015-07-19:
Argh, 2 more clicks and some scrolling at each opening... some people are already complaining that they have to choose the radio type each time :)

projectkk2glider, 2015-07-19:
Maybe we should remember the last selection and select it at the next start. Do settings work in standalone Simulator?

closed 2015-07-19 by projectkk2glider

kilrah, 2015-07-19:
That would be good. Settings should work as things like SD card folder work.

projectkk2glider, 2015-07-19:
@kilrah how about now? Pretty sleek i think :wink:

kilrah, 2015-07-19:
Perfect! No more problems even with multiple languages and the ton of related simulators :)

[image]

EDIT: seems it works on Windows, but on linux the simu still doesn't remember the last used item ;)

projectkk2glider, 2015-07-19:
simu is not in the same league. Use `simulator` executable from companion dir.

kilrah, 2015-07-19:
Yes that's what I meant.

kilrah, 2015-07-19:
oh no I know, I jsut repeated last command and it launched simulator2.1 that doesn't include that change. My bad.

commit 41d06758 (2015-07-19): Fixes #2503: Standalone simulator radio selection display problem
commit b4dc4f4b (2015-07-19): Fixes #2503: Standalone simulator radio selection display problem