opentx/opentx #2505: PlayNumber( alt/units, 6, 0 ) says "feet/second" on today's nightly

fraternl, 2015-07-19:
layNumber( alt/units, 6, 0 ) says "feet/second" on today's nightly

In https://docs.google.com/document/d/1ILh9lSW4d05qJD2rSKxep0WnJUfNCcAPCafKZeESweE/export?format=pdf

The 6 stands for "metres"

Maybe this has changed, but I would not know where to find the documentation for 2.1.x
I can't determine if it's a bug.

projectkk2glider, 2015-07-19:
Things have changed, the unit number is taken from this enum (first line has value 0, second 1, etc..):
```c
enum TelemetryUnit {
  UNIT_RAW,
  UNIT_VOLTS,
  UNIT_AMPS,
  UNIT_MILLIAMPS,
  UNIT_KTS,
  UNIT_METERS_PER_SECOND,
  UNIT_FEET_PER_SECOND,
  UNIT_KMH,
  UNIT_MPH,
  UNIT_METERS,
  UNIT_FEET,
  UNIT_CELSIUS,
  UNIT_FAHRENHEIT,
  UNIT_PERCENT,
  UNIT_MAH,
  UNIT_WATTS,
  UNIT_DBM,
  UNIT_RPMS,
  UNIT_G,
  UNIT_DEGREE,
  UNIT_MILLILITERS,
  UNIT_FLOZ,
  UNIT_HOURS,
  UNIT_MINUTES,
  UNIT_SECONDS,
  // FrSky format used for these fields, could be another format in the future
  UNIT_FIRST_VIRTUAL,
  UNIT_CELLS = UNIT_FIRST_VIRTUAL,
  UNIT_DATETIME,
  UNIT_GPS,
  UNIT_GPS_LONGITUDE,
  UNIT_GPS_LATITUDE,
  UNIT_GPS_LONGITUDE_EW,
  UNIT_GPS_LATITUDE_NS,
  UNIT_DATETIME_YEAR,
  UNIT_DATETIME_DAY_MONTH,
  UNIT_DATETIME_HOUR_MIN,
  UNIT_DATETIME_SEC
};
```

bsongis, 2015-07-19:
The doc is not up to date.

closed 2015-07-19 by bsongis