opentx/opentx #2506: [Suggestion] Change Telemetry Sensor ID Input Method

twokaff, 2015-07-19:
Currently, when adding a sensor manually on the radio, we have to scroll through the ID values until we have the correct one.  That can take a *very* long time depending on the ID value.

Consider changing the input method for the sensor IDs to be the same as used when entering names, i.e. enter each part of the ID by selecting from hex values 0-F.  Perhaps, when no value is entered it defaults to 0.

kilrah, 2015-07-19:
The system wasn't really meant to be used that way, is there really a need to enter these manually?

twokaff, 2015-07-19:
There is the option to manually add sensors.  What if the sensor has the ID F3AB?  I'll be sitting there ages waiting for the ID list to scroll to that position.  If, on the other hand, I'm not supposed to add sensors manually, then why is the option there in the first place?

It just struck me as cumbersome to having to scroll through the IDs in increments of 1, when I could just add each digit individually and be done with it in a few seconds.  That was my reasoning behind the suggestion.

I don't fully understand the new telemetry system yet, perhaps my suggestion is moot.  If that's the case feel free to close it.

bsongis, 2015-07-19:
Fair enough. I don't see why this field is modifiable.

closed 2015-07-19 by bsongis

kilrah, 2015-07-19:
Because it doesn't hurt and even if not convenient it lets you change it should you need it. I've edited the ID and instance a few times during testing.

kilrah, 2015-07-19:
Because it doesn't hurt and even if not convenient it lets you change it should you need it. I've edited the ID and instance a few times during testing. I don't see a reason to spend time making it easier, but I'm definitely not for removing the possibility either.

commit 1f06e798 (2015-07-19): Fixes #2506
commit 3ab823fb (2015-07-19): Fixes #2506
commit 668fadb7 (2015-07-20): Revert "Fixes #2506"

This reverts commit 1f06e7982aae9992c29eb96b38d6adafd88b4d07.
commit e141fe85 (2015-07-20): Revert "Fixes #2506"

This reverts commit 3ab823fb3e0f474f5aca1cf870bf83988ab99b47.