opentx/opentx #2507: 2.1 Handling Telemetry sensors with same name

pinkywafer, 2015-07-19:
If I have two Temperature sensors (for example) T1 and T2, they both default to being named "Temp".
On other options (such as setting up display for telemetry screens or using as a variable in a logical switch), they appear in the order they are listed in the sensors screen.  
If I use `getFieldInfo("Temp")` in a script, the first sensor named "Temp" is used.  Is there any way to access the second (or third etc)?

kilrah, 2015-07-19:
If the user has several sensors of the same type they will rename them to something more meaningful in their context.

pinkywafer, 2015-07-19:
Ok, thanks

kilrah, 2015-07-19:
@bsongis is it possible for a script to call another script with some parameters?
If yes now that we have file storage possibility in Lua I'd well see a generic "sensor picker" script that we could supply the same way the wizard is so everybody has it.
 
On first run and upon request the user script could call that sensor picker any required number of times with 2 parameters, one being its name (if no other way to identify the caller) and the second an item name. The picker script would create or update a file on the SD card with the caller script name and model name, and store mappings of "item name" - "sensor the user chose for it". 
Once configured the user script could retrieve the mapping when launched.

If we don't have something like this it's going to be a pain to distribute Lua scripts as they'll need editing for everybody's own case, and even worse potentially for each model (which goes against the change to a common SCRIPTS\TELEMETRY folder allowing to pick the same scripts for multiple models).

pinkywafer, 2015-07-19:
That sounds a neat solution, even if the "picker" had to be implemented in each script, it wouldn't be too bad as long as we had a function to return all field names to populate the list

mpjf01, 2015-07-19:
I assume there is a reason why the sensors with the same name are not named Temp(1), Temp(2) or similar when they are first identified?

When you change receivers and sensors for a given model from D series to X series the D sensors remain in the model definition and the common sensor names become duplicated. I assume it would be the same in reverse. I think it would be a good idea to delete the "old" sensor data in this case as it cannot be used.

kilrah, 2015-07-20:
Yes, firstly there's only space for 4 chars, then it would be a pain to keep track of the numbers that are already used, and the user will always be able to pick a more meaningful name so we let him that job, it's the whole point of having freely definable names.

Deleting a sensor kills anything linked to it (LS, telemetry screens etc) so you don't want that to be done automatically.

bikemike, 2016-12-09:
Maybe a new function should be added that returns a table of all the matching field names instead of just the first one as getFieldInfo does. But then how would you distinguish one from the next? Maybe the 4 digit telemetry id should be included in the field info. For example, 0400, 0401, 0402 for three Temp1 values. And maybe also a field type to distinguish between different types (channels, switches, globals, inputs, telemetry etc). This would be useful in the case that you name a sensor to the same name as one of the predefined names. For example, if you name a sensor ch1.

The type could also be passed into the function to filter out unwanted results. For example, getFields("Temp1",ALL) or getFields("Temp1", TELEMETRY). It could default to ALL.

kilrah, 2016-12-09:
The problem is that the "0400, 0401, 0402" are nearly never used in practice (was kind of a stupid choice by FrSky, useless).
So all Temp1 values would be received with 0400.

bikemike, 2016-12-09:
Sorry, that specific example is for something I'm working on(sending some custom data to be formatted and displayed in a lua telemetry script). All T1 sensor IDs are named Tmp1 when discovered(not Temp1 as I had written a above).

But the old TEMP1_ID will also be named Tmp1 when discovered.

#define TEMP1_ID                  0x02

kposadowsky, 2017-01-30:
Tmp1 and Tmp2 are used (or lets say abused) by cleanflight to send Lock and GPS Stats for smartport telemetry:

https://github.com/cleanflight/cleanflight/blob/master/docs/Telemetry.md#smartport-sport

I am currently writing a lua script for it. Better way would be to have all there fields seperatly.

closed 2020-07-08 by stale[bot]