opentx/opentx #2508: Firmware Simulator 2.1 remember positions?

pinkywafer, 2015-07-19:
Would it be possible for the standalone simulator to remember stick / switch positions etc?
Also, I use mode 2, but the simulator sticks always default to mode 1.ie, although the simulator configuration is saved and functions in mode 2, the wrong axis is "sprung", and I have to deselect "Hold Y" on the right stick, and select "Hold Y" on the left stick

BenZoFly, 2016-05-29:
Is someone working on this matter? I would also like to propose to add the pots to the values remembered. I use volume on pot S1 and it always revert to 0 and so I have no sound by default.

Either ticking the "remember position" remembers all positions (also toggles like "hold X") or those that needs to be remembered is selectable (like selecting stick warnings in "setup" screen)

Thanks!

kilrah, 2016-05-30:
>Is someone working on this matter? 

Most likely not, the simulator interface needs a complete overhaul anyway to adapt to the X9E and Horus controls so nothing will be done until that happens.

mpaperno, 2017-01-27:
As per #4282 (commit 6e2474081f216d255853cbec56a997d4ac8eab64)
Switch/know/slider positions are now saved per radio profile, and sticks are properly set up for the corresponding mode configured in the radio general settings.  We may also save actual stick positions later on, but not sure that's necessary or useful.

closed 2017-01-27 by mpaperno