opentx/opentx #2509: 2.1 RPM sensor read strange data

driver33, 2015-07-19:
Hi,
RPM sensor almost not working :
- Optical RPM sensor on hub connection connected together with Temp hub sensor via vario as a "translator" for S-Port telemetry.
Sensor is recognised correctly by Telemetry system.
I use it because now there is no other possibility on Fr-Sky system to get info of Glow Engine.
When it is set to show rpm it always shows 1RPM No change of the value. 
When its set to show NO-UNIT value the number shows for 0 to 28x <290.
Real RPM of my engine is from 2400-9200 (full throttle in downflight)

On 2.0.17 it worked perfectly - difference in measurements via Telemetry and manual tachometer was less than 20-30 rpm when engine was on full throttle (8200 - max rpm when plane is standing still ).

projectkk2glider, 2015-07-19:
You need to set sensor `Multiplier` value to some reasonable value. Let me see if I can found out which multiplier is right for FrSky's RPM sensor...

projectkk2glider, 2015-07-19:
That would be 30.

projectkk2glider, 2015-07-19:
@bsongis  Perhaps we could set the Multiplier to 30 when the FrSky RPM sensor is detected.

kilrah, 2015-07-19:
It should normally be 1 and OK as is. Multiplier was added only for people who have fractional ratios like heli gears, otherwise multiplier should stay 1 and you only use blades as divider. At least for smart port and that was tested and OK, for D it's supposed to be the same on paper but I'm afraid nobody has likely tested.

projectkk2glider, 2015-07-19:
This was equation in 2.0:
```c
frskyData.hub.rpm *= (uint8_t)60/(g_model.frsky.blades+2);
```
I guess (based on telemetry simulator) that in 2.1 the raw value is used directly.

projectkk2glider, 2015-07-19:
Yep, the new code in 2.1:
```c
newVal = (newVal * sensor.custom.offset) / sensor.custom.ratio;
```
The `sensor.custom.offset` is Multiplier parameter in menus for RPM sensor.

kilrah, 2015-07-19:
OK, try with the multiplier... but if that works we have a problem because it works correctly without when using a smart port RPM sensor, and I don't believe they should be different.

projectkk2glider, 2015-07-19:
That is strange, because according to the 2.0 sources the raw to rpm conversion was the same for D and smart port sensors.

driver33, 2015-07-19:
Keep in mind it worked perfectly on 2.0.X

projectkk2glider, 2015-07-19:
@kilrah has anybody actually tested smart port RPM sensor?

driver33, 2015-07-19:
I can try to test on my other Plane this sensor for electrical engines with 2 temp sensors.

projectkk2glider, 2015-07-19:
I think this is a solution https://github.com/opentx/opentx/commit/8ec29540a49a0d5486d6b457fda2402811a21467 but we must first verify if smart port works as it is.

driver33, 2015-07-19:
OK just tested this on EMAX GT2815/06 (1250KV).
Max PRM on 2 CELLS is 16600 - (no propeller - Storage loaded 3SLipo). (which seems guite correct)
BUT !!!! : Number of propeller blades is set to 1 ! not to real number of magnets in the engine as it was in 2.0.

projectkk2glider, 2015-07-19:
Did you set number of motor poles as per sensors user manual www.frsky-rc.com/download/down.php?id=159

kilrah, 2015-07-19:
> has anybody actually tested smart port RPM sensor?

Yes myself with the FrSky sensor and a signal generator. Sensor configured for 1 pole (so we enter the number of poles as "blades" on the radio), but of course you could also leave blades = 1 and set the number of poles on the sensor.

The fix is IMO however not correct, the multiplier should be 60 and should be "baked in" i.e.

> newVal = (newVal * 60 * sensor.custom.offset) / sensor.custom.ratio;

Makes no sense having an adjustment to the protocol that is visible to the user. The user should only enter values that are determined by him, not magic constants.

projectkk2glider, 2015-07-19:
I am beggining to believe that old hub and new smart sensor report different value for the same RPM. Found this comment:
>I purchased this sensor to use with a [pre-Plus] Taranis X9D running the official FrSky firmware (v1.1.02 as of this writing). The good news is I have compared the temperature probes against an accurate thermometer and found them to be right on the money. The bad news is the RPM sensor will not report the correct RPM with this radio/firmware combo. After many emails back and forth, FrSky's only "resolution" for this issue is to flash Taranis with the unofficial OpenTX v2.xx firmware. Unfortunately, I cannot report if this will indeed fix this issue since I am not willing to go there yet. So just be aware that if you have an original Taranis running official firmware, you will not get the correct RPM displayed. With correct RPM, I would rate this sensor 5 stars.

projectkk2glider, 2015-07-19:
I vote for things to stay as they are now. Users of old hub sensors must set their multiplier to 30, others are OK with defaults.

driver33, 2015-07-19:
@projectkk2glider As far as I remember I set poles inside sensor - it was 2-3 month ago. I don't use electric model since that time.

kilrah, 2015-07-19:
> Then that could be your problem in the first place, or at least part of it. If you set poles in the sensor then its output would be divided by the amount of poles you set. If you try it on a different motor you'll need strange multiplier/blade numbers to compensate for it. 
either set poles right in the sensor, or leave them as 1 and set on the radio, but avoid combinations.

driver33, 2015-07-19:
Andree I don't have problem with Electric RPM sensor it works and shows value correctly - problem is with optical HUB RPM sensor :
When HUB RPM sensor is units is set to rpm you can get only 1 value === 1. 
Without setting rmp as unit for this sensor you can get more values but they also seem to be sensless.

kilrah, 2015-07-19:
Also, again the multiplier should be 60, not 30. In 2.0 blades were shifted by 2.

kilrah, 2015-07-19:
> When HUB RPM sensor is units is set to rpm you can get only 1 value === 1. 

I understood that. But if you set multiplier to 60, do you then get the correct value or not?

driver33, 2015-07-19:
Nope I think problem lies elswhere. BTW why 30 or 60 ?? Our (european) electrical system is 60Hz so optical RMP sensor shall show 3600 (one blade) 1800 (2 blades).
To be clear I can run glow engine at home so lamp is very good calibrator.

I got this correct value in this situation : 
RMP sensor unit set to none, multiplier set to 36. 1 blade.
When RMP sensor unit is set to rpm as should be it give always only 1 value 1* multiplier (or 0 when off)

kilrah, 2015-07-19:
Because apparently FrSky decided to send multiples of 60rpm instead of the real rpm number.

driver33, 2015-07-19:
OK. Nevertheless, the HUB RPM sensor when correctly set show only 2 values 0 if plane is off, 1 if plane is on.

kilrah, 2015-07-19:
I found both a hub and a sport sensor in my bag, will try when I can and do a proper fix.

bsongis, 2015-07-19:
I would like to see this one fixed in 2.1.1, @kilrah would it be possible for you?

kilrah, 2015-07-19:
Will try tomorrow

closed 2015-07-21 by kilrah

driver33, 2015-07-21:
Confirmed - works correctly.

commit 4626550c (2015-07-21): D-type RPM sensor was broken (Fix #2509)
commit 12c111bd (2015-07-21): D-type RPM sensor was broken (Fix #2509)