opentx/opentx #2511: Mega 2560 Can't connect to radio after firmware flash

WillD7545, 2015-07-20:
After flashing firmware to 2560 SainSmart clone, companion cannot communicate with radio. Receive timeout errors.

JivaroFAD, 2015-07-20:
You must press the reset button of board just before flash.

Please read : http://jenkins.open-tx.org/MEGA2560/2_Configuration-Flash/MEGA2560_FlashProcedure.pdf

closed 2015-07-20 by JivaroFAD

WillD7545, 2015-07-20:
It writes, but when the radio boots, eeprom data is bad and it reformats.

kilrah, 2015-07-20:
Could be the usbasp is loaded with the firmware that fails to flash the m2560 eeprom correctly, common problem... Would be stupid on a board that precisely uses an m2560, but then with clones anything is possible.

JivaroFAD, 2015-07-20:
I assume you talk about usb operation, not usbasp. With usb driver based on A16U2 or A8U, you can not flash firmware and restore eeprom in the same operation.
You need to decompose each operation, with a hard reset at each time.

WillD7545, 2015-07-20:
Yes, the reset button is pressed before each operation.