opentx/opentx #2513: "Copy sensor" function on telemetry item popup

kilrah, 2015-07-20:
The system is made so that you can have multiple copies of a sensor with different attributes, but at this point you can only get to it if you enter everything manually. That's all good although cumbersome for things like Alt, but we've got a problem for a few sensor types that have hidden attributes that can't be entered manually.

projectkk2glider, 2015-07-20:
The received telemetry value is set only for the first sensor that is found, look at https://github.com/opentx/opentx/blob/cd3d473ddcbf245c392fcec744a10749a34cebba/radio/src/telemetry/telemetry.cpp#L489

So your method will not work. What will work is to create a calculated sensor that uses original value and has attributes set to your liking.

kilrah, 2015-07-20:
Read that again. I precisely changed that piece of code to be a loop (see the for()) so it would allow this use case.

projectkk2glider, 2015-07-20:
I see a return after the value is set for the first found sensor.

kilrah, 2015-07-20:
Then I need to go through the log to see who broke my work :(

bsongis, 2015-07-20:
Perhaps it's me to optimize things :)

projectkk2glider, 2015-07-20:
34d096e115a940d47509f229383efeef1ad45294

bsongis, 2015-07-20:
sorry I didn't know it would be needed to continue in the list (aka 2 sensors with the same IDs)!

commit b427fd13 (2015-07-20): Merge pull request #2523 from opentx/projectkk2glider/issue_2513_duplicate_sensors

Fixes #2513: set value of duplicated telemetry sensors
commit 39c0b79f (2015-08-01): Copy telemetry sensors (Fix #2513)
closed 2015-08-01 by kilrah