opentx/opentx #2516: Companion 2.1 Log Viewer Y-axis assignment

flarssen, 2015-07-20:
I did some testing regarding this issue: https://github.com/opentx/opentx/issues/2267
Looks like it needs some tweaking.
Data close in valuerange get multiple axis (cellx), and when all four are in use, there are no axis displayed when yet another is needed.

[image]

[image]

[image]

griebd, 2015-07-20:
Well, actually it's as expected... we add a maximum of 4 axis without scaling, if one more axis is need we fall back to 1 axis and scale all data to be visible!
There is no optimal solution, that one was the one most people though better....

griebd, 2015-07-20:
if you did see my previous comment on this, ignore it....
I will have a look on that later!

griebd, 2015-07-28:
as I thought it's related to the ranges of the data to plot.

let me explain how it's implemented, starting from an empty plot:
1. the range of the first data selected is used on the first y-axis (with a little margin at top n bottom)
2. when a second data is selected the following condition are checked and either a) or b) happens:
- the range of the new data is 30% bigger or smaller than the actual range?
- the lower value of the new range is bigger than the actual range upper value?
- the upper value of the new range is lower than the actual range lower value?
a) if any is true a new y-axis is created
b) if they are all false we plot the data with the same y-axis
3. when more data is selected the conditions above are checked for each existing y-axis and new y-axis are created when necessary until a maximum of 4 y-axis
4. if a fifth y-axis is needed we fall back to rescaling data (so all data is visible) to use only a single y-axis (with no scale as it would be nonsense anyway)

so, all is working as expected. BUT, apparently the 30% ratio (taken from thin air) is to low!

Question to @bsongis or @kilrah or @projectkk2glider (the main ones we all should be thankful): should I change this value to something reliable for this particular case? should it be user selectable? or what? opinions, suggestions and ideas are welcome (from any and everyone)!

flarssen, 2015-07-29:
Thanks for the explanation. Seeing how it works, I think you can leave it as it is, at least until we get more experience with real life data. After all, a cell will rarely have a lowest voltage that is above the range of the other cells, The only thing I can think of, is to use same Y-axis even if ranges doesn't overlap as long as the averages are close, like 10%.

bsongis, 2015-07-29:
ok we leave it as it is now, it's not bad!

closed 2015-07-29 by bsongis