opentx/opentx #2517: Compagnon 2.1 not able to flash model

blairfancy, 2015-07-20:
Hi,
Companion is not able to flash model on 9XR pro, the app freeze.

bsongis, 2015-07-22:
Reproduced

closed 2015-07-22 by bsongis

projectkk2glider, 2015-07-22:
Is this a same issue as #2402?

bsongis, 2015-07-24:
Nightly builds are there, would you check the issue is fixed?
http://downloads-21.open-tx.org/nightly/companion/

blairfancy, 2015-07-25:
Hi,
Bug was not solved with nightly build. I m not able to restore model.

Regards

Date: Fri, 24 Jul 2015 00:36:36 -0700

http://downloads-21.open-tx.org/nightly/companion/

blairfancy, 2015-07-26:
It looks like bug on V2.12 with memory allocation problem

projectkk2glider, 2015-07-26:
>It looks like bug on V2.12 with memory allocation problem 

@blairfancy  can you explain a little more, what allocation problem and which version, the 2.12 is not a valid version number.

blairfancy, 2015-07-27:
Please see issu #1793 for more information.

commit a94d0860 (2015-07-22): Fixes #2517