opentx/opentx #2519: Companion 2.1.99 crashes when I open general settings

fraternl, 2015-07-20:
Companion 2.1.99 crashes when I open general settings.
It doesn't do this when it's empty...

It does so when I open General settings after opening an eepe or reading the settings from my radio.
I just tested it using remote desktop on my machine at work and it does the same.
It seems to be related to my radio's setup, so I will attach the eepe
[image]

projectkk2glider, 2015-07-20:
Confirmed, also in 2.1. I will take a look...

bsongis, 2015-07-20:
Isn't it a compilation problem? Does the problem exist if you start from an empty build dir (not only make clean)

projectkk2glider, 2015-07-20:
@bsongis the problem is deeper than I thought.

The issue is in available sources for global function `Play Value`. The problematic are telemetry soruces which are picked from current model! On the radio it sort of works, but it might not with other model selected. In Companion it doesnt work, because model is not available when we populate CB sources!

projectkk2glider, 2015-07-20:
I think telemetry values should not be available for selection for global function `Play Value`. At least not on ARM platforms where we have the new telemetry!

bsongis, 2015-07-20:
True.

bsongis, 2015-07-20:
I had the same problem when thinking about Global Switches ...

projectkk2glider, 2015-07-20:
Will you fix it, or shall I?

bsongis, 2015-07-20:
I let you work on it, I still have a ton of emails to answer, but let me know if you cannot!

projectkk2glider, 2015-07-20:
Ok, I will try...

fraternl, 2015-07-20:
I'm getting bugs that were fixed before.
It seems as if I have an older firmware on my radio all at once.
I flashed it just now using Companion 2.1.99

fraternl, 2015-07-20:
Is this a quick fix and will it be added later?
It sort of makes it crippled.
I was so glad I could finally make that radio wide

projectkk2glider, 2015-07-20:
You can't have a global function that plays telemetry value from one of the models. Because the other models might have something else there or worse nothing. Think abut it.

fraternl, 2015-07-20:
And getting back the radio wide variables Alt and RxBat?
We already have to add 2 switches in each model for that... (I ran out of logical switches)

projectkk2glider, 2015-07-20:
With the new telemetry system there are no more any radio wide telemetry values. Even RSSI and SWR can be in different positions in different models and can have different names.

Currently I see no other solution for your problem. I agree it is perhaps more cumbersome. Maybe we will come up with some clever idea how to improve this situation.

closed 2015-07-20 by projectkk2glider

fraternl, 2015-07-20:
Link them to radio wide telemetry variables in each model and allow these to be referred to in Global Functions?

projectkk2glider, 2015-07-20:
Your idea might work. But it needs to be discussed more. Please create a new issue with your requests and ideas, this one is closed and others will miss discussion if we continue here.

bsongis, 2015-07-20:
Yes this is not a bad idea. Will use more RAM for sure though! But it would also fix the Global Switch issues I foresee.

PR #2522 (2015-07-20): Fixes #2519: removed telemetry values from available selection in Glo…
…bal Functions for "Play Value"

Closes #2519
commit 611f8511 (2015-07-20): Merge pull request #2522 from opentx/projectkk2glider/issue_2519_global_functions_play_value_fix

Fixes #2519: removed telemetry values from available selection in Glo…
commit 841f11c2 (2015-07-20): Fixes #2519: removed telemetry values from available selection in Global Functions for "Play Value"
commit f62e15c1 (2015-07-20): Fixes #2519: removed telemetry values from available selection in Global Functions for "Play Value" (ported from master)