opentx/opentx #2520: "my" battery watchdog says battery critical when the model binds

fraternl, 2015-07-20:
In 2.1.x we have to take care of the RxBat ourselves which is fine...
So I created these 2 switches and 2 special functions

L1 ( RxBt < 3.7 ) AND !L2 Duration (2s)
L2 RxBt < 3.6 Duration (2s)

SF1 L2  Play Track batlow 60s --- 
SF2 L1  Play Track batcrit 15s ---

AFAIK know they work fine, but when the receiver binds I will get 1 voice alert "battery critical" (SF2)
Nomally I would solve this with the !1x that will mute the voice alert on binding, but I want it to be repetitive alarm.

Maybe I should set it up diiferently?

projectkk2glider, 2015-07-20:
Use `delay` instead of `duration` for both switches.

projectkk2glider, 2015-07-20:
Please reopen if it still doesn't work after trying my suggestion.

closed 2015-07-20 by projectkk2glider

fraternl, 2015-07-20:
Can't test it.... I only brought my radio with me (models/receivers at home).
I believe you, though.

projectkk2glider, 2015-07-20:
You can test in Companion simulator using telemetry simulator (press F4 key while in simulation window)