opentx/opentx #2521: Companion 2.1.99 imports SF "Reset Alt" as "Reset ---"

fraternl, 2015-07-20:
Companion 2.1.99 imports SF "Reset Alt" as "Reset ---"
I had this in 2.0.99, but then it was impossible to correct it in Companion.
It was then not possible to make a "Reset Alt"

Now, although it's correct in the radio, it will import that setting as "Reset ---"

I believe it worked before, but can't be 100% sure

projectkk2glider, 2015-07-20:
@fraternl  Just a side note. I recommend you use the `master` branch for your testing, which now contains 2.1.x development. The `next` branch is reserved for 2.2.x.

fraternl, 2015-07-20:
I'm using nightly builds. Not compiling myself.
How do I get a proper firmware on my radio again?

projectkk2glider, 2015-07-20:
Use these night-lies http://jenkins.open-tx.org/nightly-stable/ these are 2.1.x

bsongis, 2015-07-20:
Right, I just renamed the nightly builds. Currently we focus on 2.1.X, I would like we release the 2.1.1 this week.

fraternl, 2015-07-20:
But isn't this critical enough to be in 2.1.1?

When I write it back to the radio the setting is luckily preserved.

bsongis, 2015-07-20:
Yes if the bug is in 2.1.0 it must be fixed in 2.1.1 
You lost me with 2.1.99

fraternl, 2015-07-21:
I am now using the binaries in the nightlies.
Will test again this evening for some stuff...

bsongis, 2015-07-21:
Anyone reproduces this issue? Here it works ..
I only see an issue with Reset in Global Functions
The same than #2519

kilrah, 2015-07-21:
Works fine both on 2.1.0 release and current master.

I close it, reopen with precise info if you have some to provide.

closed 2015-07-21 by kilrah

fraternl, 2015-07-22:
Didn't see it was closed without a change.
I will check it again this evening and give some more info.
I'm sure it's still in there "Print model" gave an entry in red as well.....