opentx/opentx #2524: Installed 2,1,1 from nightly-stable and am getting several bugs back that were fixed last weekend

fraternl, 2015-07-20:
I just installed Companion 2.1.1 from http://jenkins.open-tx.org/nightly-stable/2015_07_20/

I let it download opentx-taranisplus-timer3-noheli-lua-mixersmon-en-2.1.0.bin and flashed it on my radio.
I have several bugs that I saw fixed before...

I chose a different option (faichoice) to force a new build on your servers (that is how it works, doesn't it?) and I got opentx-taranisplus-timer3-noheli-lua-mixersmon-faichoice-en-2.1.0.bin and flashed it.

Still old bugs...

I am probably doing something wrong....
Would like to continue testing. Do you want me to?

bsongis, 2015-07-20:
Which bugs? Currently there is no system to have nightly companion linked to nightly built firmwares. You have to download them separately.

bsongis, 2015-07-20:
Yes we need your tests. Just try to have the explanations as good as possible. We try to do our best, but we have to switch quickly from one topic to another and sometimes the context is missing. Yes, videos, screenshots are often the best option to explain. At least with me.

projectkk2glider, 2015-07-20:
When you download the new firmware with companion 2.1 (even if you use nightly), your firmware gets built from the released sources and thus does not contain any fixes since the last public release!

Your only choice is to use the firmware in the same folder. That one is build from current sources, but has the default options selected (which might not be what you want). Or build it yourself.

fraternl, 2015-07-20:
The easiest option would be to use those binaries that are in the nightlies.
But if they don't have LUA I will not...
Worked hard to get some scripts going again in 2.1

bsongis, 2015-07-20:
These nightly builds are static, you need a dynamic system, that's the problem. It's feasible, just needs more time.

projectkk2glider, 2015-07-20:
The nightly's have : HELI, LUA and GVARS

fraternl, 2015-07-20:
OK... fine by me..
Just loaded it and saw those bugs fixed again....
Will now check some more and close the issues I made today

closed 2015-07-20 by fraternl

projectkk2glider, 2015-07-20:
Today's fixes are not yet in the last nightly build!