opentx/opentx #2525: Telemetry multiple sensors based on DIY boards problem

Datospel, 2015-07-20:
Hi!

Found the following problem that would matter for anyone who uses any kind of data collecting DIY project that packs data to SPort protocol. My case: I have a Teensy board that is connected to Naza GPS and collects data (GPS coords, heading, etc) packs it to SPORT and sends it though X8R to Taranis PLUS. 

Taranis discrovers all sensors and data coming from this board upon first connection. Finally, I get many similar Temp (4), Vfas(4), Vspd (3) etc values. Therefore, further identification of those values is very hard - I just cannot tell whic stand for what. I understand, that this is not a problem of OpenTX developers to identify the data coming from such boards... instead, this is a problem of how DIY board developers pack this data and under which labels. Anyway, when everything is connected and working (still not identified properly) Taranis starts to show a popup saying "All telemetry slots are full". This popup cannot be closed and removed from the screen. It is constantly reproduced. 

Maybe, one of the soulutions would be to giva a possibility to a user to choose which sensors (among discovered ones) to switch off. This would give a possibility to switch a sensor or value one by one and have a time to identify each one without fighting popups and other issues that have to do with some wrong id's or data overflow....

Thank you

projectkk2glider, 2015-07-20:
Your descriptions leads me to believe that the telemetry generator has as bug which then causes Taranis to recognize the new sensors repeatedly until all available slots (32) are full. 

Which project exactly you have there and which version?

bsongis, 2015-07-20:
Your sensor was replying to all polled IDs ...
I already wrote a fix for your Teensy project:
https://github.com/bsongis/MavLink_FrSkySPort

closed 2015-07-20 by bsongis

Datospel, 2015-07-20:
I have this project http://www.rcgroups.com/forums/showthread.php?t=2148476 Teensy to Naza to Taranis. 

bsongis - thank you, but I have another one

bsongis, 2015-07-20:
Any link to the code so that I can check the S.PORT polling loop?

Datospel, 2015-07-20:
it is not opensource. autjor keeps it closed for public...

Datospel, 2015-07-20:
there are similar not opensource projects that are being sold... that is why I propose to make "sensor off" available

bsongis, 2015-07-20:
So that OpenTX fixes wrong S.PORT implementations?

Datospel, 2015-07-20:
no, for a user to decide which sensors and data to show and which to avoid. or to be able to identify each one in steps instead of getting all the list populated with some unrecognized values....

projectkk2glider, 2015-07-20:
I would not change our implementation to accommodate such bad project. 

Perhaps only beneficial change would be to get rid of the warning message when all sensor slots are used up and new sensor is found. But then again some other user will complain about missing warning.

Datospel, 2015-07-20:
maybe.... I agree that the problem is in the SPORT implementation of the specific project... Anyway, what about the case when I disconnect a sensor - why it should stay present on the model if I wouldn't use it anymore? Isn't it more logical to give a pisiibility to switch it off or delete it or to rediscover all sensors?

bsongis, 2015-07-20:
You need it is present in the model, but "switched off". Did you calculate how many sensors would be in such state with a wrong implementation of the S.PORT protocol?

Datospel, 2015-07-20:
with the wrong implementation - it might not help at all. but in case if I remove a normally working sensor - switching off would be a good choice

kilrah, 2015-07-20:
> Anyway, what about the case when I disconnect a sensor - why it should stay present on the model if I wouldn't use it anymore?

Because if they were deleted automatically anytime you powered up the receiver you'd lose all your settings (logical switches, alams, telemetry screens etc) related to some values because some sensors can take a few seconds to start sending info making them appear as "not present", same if you unplugged a sensor by mistake. 

I.e. if you decide to remove a sensor from your model take the 10 seconds that are required to delete the related sensors (that functionality is there as opposed to what you imply).

And I agree that it would be nonsense to spend time doing work to fix some third party's errors, especially if they keep their thing closed-source. Send them messages telling them to fix their mess and comply with the protocol or refund you.
The "switching off" wouldn't even be an option in that case since you'd literally have to "switch off" hundreds of sensors to stop the flood from the rogue sensor.

EmanuelStassar, 2015-07-21:
Hi Bertrand,

I just uploaded your Teensy code to my copter and get different sensors then are mentioned in the Arduino files. Please check attached screenshot of the sensors that are registered. There is no Cell or Cells for instance. I hope you can point me into the right direction to fix this.
[image]

kilrah, 2015-07-21:
Is there a cell sensor in your setup? If not sending those values makes no sense so they'd have been disabled..

bsongis, 2015-07-21:
I only fixed the existing Teensy code. Now it's easy to add a new sensor inside the table. I am ok to open a pull request somewhere, just tell-me which project has the lead and I will do. I am ok to help but I cannot manage to support a new project in my spare time!

EmanuelStassar, 2015-07-21:
I understand. Unfortunately I don't know which project has the lead on this and I am no good at coding myself.

WZ9V, 2015-08-19:
I just got a quad setup with TauLabs and it's doing the the same thing.  Filling all the slots with duplicate values.  I'm assuming this means the Taulabs FC code is doing something wrong in communicating the telemetry to the X8R S.Port is that correct?