opentx/opentx #2526: Improve our build system to allow nightly built companion to download nightly released firmwares

bsongis, 2015-07-20:
Firmwares would continue to be built on the fly.

Idea taken from #2524. We really need it, it would be easier to understand for our beta testers, and we would have more feedback.

We will also need to have something to differentiate nightly builds, perhaps 2.1.1A + the date

Any idea welcome.

bsongis, 2015-07-20:
It would get rid of these 2.0.99 or 2.1.99 pseudo-versions

projectkk2glider, 2015-07-20:
Good idea :+1: 

The version should include build date (at least day and month) that would be enough I think.

bsongis, 2015-07-20:
My problem comes when we release.
I think we need something to differentiate 2.1.0 (16 of july, beta testers) from 2.1.0 (17 of july, release date)
That's why I would call the 1st: 2.1.0A (alpha) or 2.1.0B (beta)

projectkk2glider, 2015-07-20:
What if you use:
* 2.1.0 last stable release
 * 2.1.0-1 - 2.1.0-XXX stable nightly builds each day the last number is incremented
 * then release 2.1.1
 * now nighlies become 2.1.1-1 to 2.1.1-XXX

bsongis, 2015-07-20:
Yes it does the job, that's fine with me.

rdeanchurch, 2015-07-21:
Date is easier to track than just incremented  -xxxx for testers.
and requires no more letters.
Month Day   -mmdd

bsongis, 2015-07-21:
@projectkk2glider How would you name the current 2.2 (next branch)

projectkk2glider, 2015-07-22:
@bsongis  Proposed numbering/naming scheme:

Current stable is 2.1.x:
 * each release with a simple number like `2.1.x` is official release
 * each nightly build had an added suffix, for example lets look a current situation:
   * `2.1.0` is released
   * `2.1.1` is not released (still in development)
   * nightly builds for future `2.1.1` are named `2.1.1-ddmm`  (the dd being day of build, the mm being month of build). Alternatively we could use build number `2.1.1.nn` where `nn` would increment after each build. But using date/month encoding is more descriptive (but only one build per day is possible)

 * after a while we decide to release `2.1.1`. Immediately after that the master branch becomes `2.1.2` and nightly builds have version `2.1.2-ddmm`.

* Same analogy goes for 2.2 branch. Until the first release we use `2.2.0-ddmm` for nighly builds, when ready we release `2.2.0`.

I think that using `2.1.99` for the upcoming `2.2.0` is not appropriate as it is too similar to `2.1.x` branch and creates conflicts when installing different OpenTX branches on the same computer.

I've seen you already prepared makefile for the daily suffix, which is excellent. Pleas comment on my ideas. When we agree on the numbering scheme, I will create a wiki page describing it.

kilrah, 2015-07-22:
mmdd instead of ddmm for chronological sorting ;)

projectkk2glider, 2015-07-22:
>mmdd instead of ddmm

Yes, good catch. Much better. 

Use this cmdline to get it:
```
$ date "+%m%d"
0722
```

bsongis, 2015-07-22:
Well I think I will use the build index for now, I am on it.

bsongis, 2015-07-24:
So...

Nightly builds are there:
http://downloads-21.open-tx.org/nightly/companion/

In the applications parameters you can change the checkboxes to enable nightly builds for Companion and/or OpenTX Firmware. If you do so, the binaries you will download from Companion will be stamped i.e. 2.1.1N199 (which means that it's the nightly build number 199)

Once enabled you will be notified each time a new build is available (both Companion and Firmware). And of course you can return to the "normal" builds when you want.

I hope it will be easier for the users to help us to track bugs before the releases!

bsongis, 2015-07-24:
I forgot ... all commiters are allowed to launch a not-so-nightly build if they need to ask somebody to test their mod. This numbering system does allow us to have as many builds as we want per day.

bsongis, 2015-07-24:
Don't hesitate to spread this information about those nightly builds. I only updated the link on the open-tx.org website.

fraternl, 2015-07-24:
I just downloaded and installed companionInstall_2.1.1N390.exe
I expected it to download me a 2.1.x nightly, but it proposed to download 16-07-2015 firmware.
I didn't test if it was not a nightly build (on this system at work I can't use Companion).....

Can you tell me where I'm wrong.
I will be flying again tomorrow and would like to do so with the latest fixes.
Another club member just has his Taranis and I decided for him he should start with 2.1.x
He's relying on me for instructions.

fraternl, 2015-07-24:
Sorry...  should have read your post...

bsongis, 2015-07-24:
Yes ...

fraternl, 2015-07-24:
I'm the dive-in-kind-of-person   :D

closed 2015-07-24 by bsongis