opentx/opentx #2527: Cell minimum doesn't work as a variable for Logical switches

Datospel, 2015-07-20:
Hi!

I've created a telemetry value with the name MinC which is calculated ---> cels ---> minimum. It shows minimum cell voltage correctly.

I've tried to create a Logical Switch:

L1 ---> a<x ---> MinC ---> 3.50V

and than Special Function:

play track ---> battery minimum

But, L1 is active always no matter whether MinC is less, equal or more than 3.50V. Special function is also plays the chosen audio.  

________________________________________________________________________________
Another issue, creating telemetry value like MinC described above works perfect when done right on Taranis. But shows some strange values if it was created in Companion and uploaded to Taranis.

projectkk2glider, 2015-07-20:
Is this coming from the same sensor as #2525?

Datospel, 2015-07-20:
no, it is coming from FrSKY FLVSS. It works perfect...

bsongis, 2015-07-20:
Needs to be checked on FLVSS before 2.1.1

bsongis, 2015-07-20:
How many elements does your battery have?

Datospel, 2015-07-20:
Wait, I'll provide a screenshots

Datospel, 2015-07-20:
6 cell pack

bsongis, 2015-07-20:
Perfect with a screenshot, thanks!

Datospel, 2015-07-20:
[image] shows that MinC is configured right and shows 4.15V. 

[image] shows that MinC is used in L1 as a<x where "a" is MinC and "x" is 3.50V (and L2 is used for 3.35V)  BUT L1 and L2 are active (bold) despite the 4.15V reading.

Datospel, 2015-07-20:
[image] this is how MinC is configured

bsongis, 2015-07-20:
Perhaps it means that all "calculated" sensors have this issue?

bsongis, 2015-07-20:
May I have your EEPROM?

Datospel, 2015-07-20:
I didn't try other "calculated" sensors... But I could check it tomorrow. One issue is that it doesn't work right in Logical switches (and maybe in other places) but I also noticed that creating this "calculated" sensors in Companion and uploading them in Taranis = some strange readings (I've got "4" instead of "4.13")

Datospel, 2015-07-20:
wait, I'll upload it here...

bsongis, 2015-07-20:
You can modify the precision. If you select 2 digits, you will have your 4.13

Datospel, 2015-07-20:
when I've created the same "calculated" sensor in Taranis instead of Companion - the reading was right...

bsongis, 2015-07-20:
The Taranis knows his sensors, while Companion cannot.

Datospel, 2015-07-20:
But I've done this after connecting TX to RX and sensor discovery + uploaded models file to Companion and than tried to create calculated sensor. And got "4" instead of "4.15"

Here is my EEPROM file http://www.ex.ua/420052954948

bsongis, 2015-07-20:
In your EEPROM L1 and L2 are comparing Cels, not MinC

Datospel, 2015-07-20:
Oh. Yes, sorry. I've changed to Cels right after I've discovered that MinC doesn't work for Logical Switches. I forgot about this. I prefer my setup to be ready for flight. So, the problem is still here.

Datospel, 2015-07-20:
I can change it back but would it help you?

Datospel, 2015-07-20:
http://www.ex.ua/104851529420 i've changed it back. checked again - still no luck.

bsongis, 2015-07-20:
Is it the same with Max or individual cells?

Datospel, 2015-07-20:
Can't check it right now. I would check it tomorrow and post a reply for other calculated values here. '

kilrah, 2015-07-20:
I'd think this is a duplicate of #2479

bsongis, 2015-07-21:
@Datospel Then would you check with one firmware listed on this page:
http://jenkins.open-tx.org/nightly-stable/2015_07_21/
You won't find the exact firmware which will match your need, but we will know if the issue is fixed or not
Thanks

bsongis, 2015-07-21:
I quickly tested this morning, no problem on my setup, I didn't wait the cells values change though, just the logical switch. I will spend more time on it tonight if needed.

bsongis, 2015-07-21:
Any news on this one?

kilrah, 2015-07-21:
All good for me, a logical switch using a calculated minimum cell sensor works fine here. Creating the sensor (and thus associated LS) in companion 2.0.1 doesn't work but that's due to #2479 and it's fine on a 2.1.1 build.

closed 2015-07-21 by kilrah

bsongis, 2015-07-21:
I just spent more time on this one, and everything works as it should.

Datospel, 2015-07-21:
Had no time to conduct my checks today.... So, you want to say that 2.1 is working fine and there is something wrong with my setup or you are saying that it is fixed in 2.1.1?

kilrah, 2015-07-21:
There was one issue in companion that was fixed before your report and will be on 2.1.1. We could not find any problem when setting things up on the radio, so if you're sure your issue appears with the current 2.1.0 release firmware please explain exactly how to reproduce it coming from a new model.

Datospel, 2015-07-21:
ok. I'll find a time tomorrow to re-check and reproduce this issue and would make a video of the process.

Datospel, 2015-07-22:
Today, I've spent enough time to understand that the problem exists on my setup. Moreover, any calculated sensor doesn't act as it should when used as a trigger for a Logical Switch. So, let me describe as much as I can:

Given:

Taranis PLUS, rev.A 
Initial OpenTX ver. 2.0.13
Inner XJT Module
X8R receiver
FLVSS sensor
6-cell battery setup

Initial EEPROM http://www.ex.ua/951347431421

Process:

Upgraded Taranis OpenTX firmware from 2.0.13 to 2.1 using DFU (switched off) mode with Companion 2.1. The process went fine, no alerts. Took out USB cable, switched on Taranis and got models converted into 2.1 format. Uploaded models file to Companion 2.1 from Tarnis and checked its setiings. Everything worked fine. 

I've chosen Tarot960 model in Taranis and powered up my physical models with FLVSS, Normal Precision Vario and Teensy project connected to X8R receiver. Taranis populated a list of sensors on the telemetry page of the model. Then I've switched off Teensy from the receiver just to work a bit only on FLVSS and Vario readings and to avoid "all telemetry slots are ..." popup. 

I've created a new calculated sensor with a name MinC like this https://cloud.githubusercontent.com/assets/10998710/8786645/0ed64316-2f38-11e5-8d5d-b7eaa998b31a.jpg  and it shows normal and expected readings of current minimum cell.

Than I've tried to create Logical Switch L1 with a<x where V1 is MinC and V2 is 3.50V But as soon as I've created it - it became bold (active) although the reading from the minimum cell is more than 4.0V 

https://cloud.githubusercontent.com/assets/10998710/8786502/296a3cba-2f37-11e5-956b-ba90c1a7f98d.jpg

https://cloud.githubusercontent.com/assets/10998710/8786481/03eed27a-2f37-11e5-943a-2b9ddfff4831.jpg

I've tried to create another calculated sensor but choosing max cell for this time. Max cell was showing 4.15V. I've created L2 switch with a<x where V1 is Max cell and V2 is 4.0V - but the switch became active right away. 

Finally, I've tried all those operations (firmware change back and formward, models check and first initializations of telemetry sensor) 3 times and every time I've got the same result. I am not posting any new screenshots cause they would be the same. 

My final EEPROM http://www.ex.ua/104851529420

Maybe, there is such a problem when we skip from 2.0.13 to 2.1?