opentx/opentx #2529: Suggestion: Make model's telemetry variables available radio-wide so they can be referred to by Global Functions and Global switches

fraternl, 2015-07-20:
A model's telemetry variable is created on the fly and may get different names.

Global variables and switches can't refer to them directly as they may not exist.
This behaviour was changed in https://github.com/opentx/opentx/pull/2522

In the model they could be assigned to radio-wide slots
Some of them could be filled with the telemetry values that are standard for some equipment (RSSI, SWR, RxBat, Altitude, Vspd) the rest maybe abstract.

kilrah, 2015-07-20:
To me this is not possible as IDs depend on hardware and may not match between models. 
Nothing from a given model should be available or usable in anything global.

fraternl, 2015-07-22:
Creating this ticket was suggestion made by projectkk2glider & bsongis so we can discuss how to make it possible to have a way to handle these values radio-wide.

Before we had RSSI radio-wide although it was not supported by all receivers and the same goes for the battery. I understand it is a bit different now it's implemented differently, but it would be very sad if we cripple the global functions and the global switches before it is implemented.

kilrah, 2015-07-22:
Problem 1 is that different receivers report battery voltage in different ways. A D4R-II, D8R, X8R and X4R  are all seen differently by OpenTX, and sometimes configured differently. As soon as you've got a mix of receivers it becomes a mess. This could be solved by adding a field like Voltage Source where you select which of your telemetry sensors is the battery voltage. BUT then:

Problem 2 is that unless all your models use the same power configuration a global alarm makes no sense anymore, any model that's wired up differently or needs a different threshold would require you to disable global functions for it - and you'll lose all the other things like global throttle cut or volume at the same time. 

To me telemetry alarms are really too model-specific to configure globally.

fraternl, 2015-07-22:
In problem 2 you give an example for global switches. We don't have these yet.
I was using a global function that was triggered by a physical switch and it was only a "play value"

As things are I have already too few logical switches.
The new system forces me to use even more of them.

Memory has been given as an argument for limiting the switches, but if these were doubled to 64 switches, would this increase the memory usage of an almost empty model proportionally.

I would love to trade in the amount of models if I can have more switches.
If it wouldn't cost much more memory for those who don't, we have only to gain.

I understand that it would also increase the memory usage of the model in use....

fraternl, 2015-08-24:
Without looking at the restraints to implement this, the concept would work to have a limited set of radio-wide telemetry slots like RSSI, Alt, RxBat.... together with the possibility to assign inside the model a telemetry slot to such a radio telemetry slot it would become a bridge between radio-wide/model-wide

By default the slots could get auto-assigned if the radio-wide slots are empty.

fraternl, 2016-04-11:
By using reserved names for Telemetry sensors you can make these available for radiowide functions and later radiowide switches.
For instance if the sensor is named "Curr" in that model, it will be possible to reference it as "Current"

closed 2016-11-27 by fraternl