opentx/opentx #2530: OpenTX 2.1 fails to detect VFAS

flarssen, 2015-07-20:
I have a FAS40 with FLVS-01 connected to D series RX, and I get no VFAS sensor. I do get "Curr" and "Cels" sensors that works, displaying the FAS current and FLVS-01 total voltage.
With my DIY sensor, sending voltage as ID 0x39, VFAS is detected but reads 0.98V instead of 9.8V

kilrah, 2015-07-20:
Duplicate of #2518.

closed 2015-07-20 by kilrah

flarssen, 2015-07-21:
OK, but I didn't see anything related to my ID 0x39 problem mentioned above. 1/100V instead of 1/10V is fine with me as I can just skip dividing my data by 10, but I guess it might cause problems for others.
Setting the ratio, whatever that is, to 25.7 gives correct readings. Is this the way to do it?

kilrah, 2015-07-21:
VFAS is not an official field for D series. If I remember well it was added to allow for 2 decimals, an advantage you're losing by dividing by 10.

And again some missing relevant information - what is your DIY sensor based on?

flarssen, 2015-07-21:
VFAS is documented as 0x3A and 0x3B in the FrSkys HUB protocol pdf. What's not mentioned, is that voltage should be given as half of the actual voltage. At least that's my conclusion after monitoring the FAS40 output yesterday. For a 3S lipo, I got 5E 3A 5 0 5E 5E 3B 7 0 5E = 5.7V, 10.92V on multimeter.
Probably the reason why I couldn't make sense of the 0x3A/0x3B, and choose 0x39 instead, after reading a suggestion somewhere. 

My sensor is based on the data units for the Hyperion EMeter (MDU/LDU/RDU). They give voltage with two decimals. In order to get it displayed correctly on Taranis when using ID 0x39, I had to divide this by 10, at least with 2.0.15. With 2,1, this is not necessary anymore, so something has changed. If it now works as planned, that's fine.

kilrah, 2015-07-21:
3A/3B indeed has some kind of weird ratio added to it, easy to find in opentx sources.

projectkk2glider, 2015-07-21:
The 0x3A and 0x3B has as resolution of 0.2 volts. The custom added 0x39 has a resolution of 0.1 volts.

bsongis, 2015-07-21:
Perhaps we can fix that in the sources, what do you think?

kilrah, 2015-07-21:
I don't think we can, it's due to that stupid ratio. Basically the sensor divides the value by "about 2" before sending it, so we have to multiply it by "about 2" to get the right value back and lose "about half" of the resolution as if I remember well the sensor only sends one decimal.

projectkk2glider, 2015-07-21:
I will fix this while fixing #2531

flarssen, 2015-07-21:
I changed the code of my sensor and got the full 0.01v resolution that the data source provides.
Would it be possible to switch between 0.1v and 0.01v based on the selected resolution in the telemetry page, instead of fixing it back to 0.1.

PR #2536 (2015-07-21): Fixes #2531: VFAS id 0x39: was 10x to small
Fixes #2530: VFAS id 0x3A and 0x3B: wrong calculation of value

Closes #2530, closes #2531 

I've only tested #2531 on my setup and its is fixed.

The #2530 issue was **not tested** I don't have that sensor. The formula was taken from 2.0 sources and multiplied by 10.
commit 6fffd7fb (2015-07-21): Fixes #2531: VFAS id 0x39: was 10x to small
Fixes #2530: VFAS id 0x3A and 0x3B: wrong calculation of value