opentx/opentx #2533: Sensor menu has two lines with "name"

projectkk2glider, 2015-07-21:
One can find sensor name at the top and at the bottom of the sensor setup menu. The bottom one is I think a bug.

bsongis, 2015-07-21:
A screenshot? I don't see it!!!

projectkk2glider, 2015-07-21:
[image]
[image]

projectkk2glider, 2015-07-21:
Hmm, it is not present on the simu. Only on my radio (self compiled). Weird.

bsongis, 2015-07-21:
It should be "Logs" with a checkbox. It is ok on my radio.

projectkk2glider, 2015-07-21:
Yes I see it in the simu. Why is there name on my radio, I have no clue. I can even edit the name there and it works!. I will try to rebuild....

bsongis, 2015-07-21:
Perhaps it hides something really bad.

bsongis, 2015-07-21:
Who added the ONLYPOSITIVE?

projectkk2glider, 2015-07-21:
Me :grinning:  am I to blame?

bsongis, 2015-07-21:
No, me who still didn't rewrite this menu system ... all ideas welcome!

projectkk2glider, 2015-07-21:
Uff, the menu system is so deeply embedded... it would be a big undertaking.

bsongis, 2015-07-21:
Yes, but if you find a good inspiration source, I am interested.

projectkk2glider, 2015-07-21:
Tested and confirmed, the fix is good. :+1: 

I am a little ashamed to admit :blush: , but today was the first time I used 2.1 on my working radio and hooked up to a model. At least I found a couple of bugs.

bsongis, 2015-07-21:
And do you like 2.1?

projectkk2glider, 2015-07-21:
Yes of course. For my relatively simple multi-rotor model the conversion process (on the radio) was flawless. I have still to verify more complex models (F5X sailplanes). But looking gooood.

griebd, 2015-07-25:
I may be wrong on that, but I think the problem is still there but in a different case...

I have a self compiled OpenTX from commit 2b25071 for Taranis Plus (make PCB=TARANIS PCBREV=REVPLUS HELI=YES GVARS=YES MIXERS_MONITOR=YES TEMPLATES=NO).

Have a model configured with X6R receiver, FLVSS and RPM sensors.

In TELEMETRY screen If I add a new sensor with the following configuration I get a 'name' line at the end.
Type: Calculated
Formula: Cell

The 'name' line appears as soon as I select Cell in Formula!

kilrah, 2015-07-25:
Just tried with current master, same command line and same sensors, can't reproduce... did you make clean before?

griebd, 2015-07-25:
yes, I did... but to be sure I updated to the current master, make clean and same make again... and updated the radio with the new file, of course. ;)

still the same, 'name' line at the end! can't get to it and when I update the name in the first line the last gets updated too...

would a backup of my model help?

kilrah, 2015-07-25:
maybe yes... but what would be even better is the exact steps to reveal the bug starting from a new blank model.

griebd, 2015-07-25:
ok, did try a new model on radio and on simulator:
1. create new model
2. go to telemetry screen
3. add a new sensor with: name: a type: custom id: 0300 0 unit: v precision: prec2
4. add a new sensor with: name: b type: calculated formula: cell
5. and then in that moment you have the error that is different on radio and on simulator
a) on radio the last line is 'name' as described before
b) on simulator the last line ('logs') is duplicated, both get selected together and I can't mark it

bsongis, 2015-07-25:
I don't reproduce on simu
Are you on latest nightly builds? You say unit is 'v', it should be 'V' in the latest nightly builds, do you confirm?

kilrah, 2015-07-25:
OK... I checked out that precise commit this time and built both firmware and simulator on Windows (firmware with the "approved" gcc 4.7-2014q2), flashed to my T+.

Following those exact instructions I don't get any issue neither on the radio nor on the simu (and both behave the same). 

[image]

[image]

[image]

> 3. add a new sensor with: name: a type: custom id: 0300 0 unit: v precision: prec2

Note that this will not create a valid Cells sensor, you won't be able to select it in the "b" sensor as a source. At this point a Cell sensor can not be entered manually, it has to be autodiscovered. It uses a special "virtual unit" that allows handling the special case it is but is not available for manual selection.

griebd, 2015-07-25:
about your note at the end, no problems, I did this way just to see the error appear on my model the sensor is auto-discovered...

about the problem, I will dive it the code myself and see what is going on, maybe a problem with compiler versions... (was reluctant to do so because it's the first time I'll look at the radio code itself)

if I find the problem I'll made a pull request! and as this apparently is a problem of mine, lets close this issue again. ;)

griebd, 2015-07-25:
@bsongis, yes, it's V, I mostly always forget to use 'caps'... :wink:

anyway, just discovered why you are not seen the issue on simulator!
I was building companion for daily use, so I used 'cmake -DCMAKE_BUILD_TYPE=Release' by default...
if I build a Debug version I have no problems in companion simulator too. same if I avoid -DCMAKE_BUILD_TYPE altogether!

I probably shouldn't have used -DCMAKE_BUILD_TYPE=Release, right?

but that is still not the case for the radio itself, no cmake in that case. but still a different version of gcc, I will try your 'official' version later.

projectkk2glider, 2015-07-26:
That is a rather clear indication that some variable is uninitialized.

projectkk2glider, 2015-07-26:
Found one problem on simu: the Cells sensor :
[image]

projectkk2glider, 2015-07-26:
Found one bug, fix coming soon...

closed 2015-07-26 by bsongis

griebd, 2015-07-26:
@projectkk2glider, thanks!!!! checked both in simulator and in radio, they are both correct now!!!

by the way, working with:
gcc (GCC) 4.8.2
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]

compiled with:
make PCB=TARANIS PCBREV=REVPLUS HELI=YES GVARS=YES MIXERS_MONITOR=YES TEMPLATES=NO
cmake -DCMAKE_BUILD_TYPE=Release ../companion/src/
make

kilrah, 2015-07-26:
I don't know if the situation has evolved on either side but we used to have problems with gcc 4.8 and recommended not to use it.

griebd, 2015-07-26:
talking about the firmware, what kinds of problems?
I have it compiled with no errors or warnings.... and so far the only problem I have noticed was the one reported and fixed above!
I did see the recommendation to use 4.7, but 4.8 was already available and laziness convinced me to use it!
anyway, will try to keep using it (until I have a reason to change) and if you need some tests done on it, please tell me! I'll gladly help.

bsongis, 2015-07-26:
Yes there were problems with 4.8 when we compiled in -O2
Now we use -Os (and we also fixed a bug in CoOS) but we didn't switch to 4.8
If you confirm it works perhaps we will change our mind for 2.1.2

commit 4509b1a2 (2015-07-21): Fixes #2533
commit 877b96b6 (2015-07-21): Fixes #2533
PR #2571 (2015-07-26): Fixes #2533: menu was wrong if all items on the end were hidden
This fixes problem on the sensor menu.  Closes #2533.

**But other menus** uses the same kind of code and could have the same issue.
commit 09d6fa83 (2015-07-26): Merge pull request #2571 from opentx/projectkk2glider/issue_2533

Fixes #2533: menu was wrong if all items on the end were hidden