opentx/opentx #2538: Allow sensors which have a bad S.PORT implementation

bsongis, 2015-07-22:
Such as EagleTree one (closed source). I don't like that too much, but perhaps we are able to avoid that a lot of those sensors are trashed...

I see a simple way: add a boolean such as "Only unique IDs" or "Bad sensors", we could avoid the check of the ID and consider that 2 sensors with the same type can be merged (they are in fact one)

What do you think?

projectkk2glider, 2015-07-22:
I have an even simpler idea: when the special new option (model wide) is ticked we would just overwrite/set incoming sensor instance number to the same value (lets say 0) before processing as usual.

kilrah, 2015-07-22:
I have used Eagletree products in the past, including reporting bugs and asking for features and they are good and quick to correct things once they're aware of it. Nothing to do on our side, once users complain at the right place (I've made a post on RCG) it will likely be fixed as quicky as we would.

bsongis, 2015-07-27:
Checkbox added.

closed 2015-07-27 by bsongis