opentx/opentx #2541: Companion 2.1.1 does not correctly import SF "Reset Alt"

fraternl, 2015-07-22:
Actually ticket: https://github.com/opentx/opentx/issues/2521

Companion 2.1.1 (22 Jul 2015)
Firmware 22 Jul 2015
http://jenkins.open-tx.org/nightly-stable/2015_07_22/

Special Function 15 which contains "reset Alt" will get a "reset ---" if it is imported by Companion.
The setting is not lost when I write it back to the radio.
Companion is able to set a "reset Alt" (not possible 2 weeks ago)

In print model this Special Function has an error in red  (see screenshots)

[image]

[image]
[image]

fraternl, 2015-07-22:
[image]

fraternl, 2015-07-22:
To make sure you see it I used the yellow marker:
[image]

kilrah, 2015-07-22:
Aha, much better when we've got details to work with!

Your EEPROM has empty slots before Alt, and that is what is revealing the bug.

fraternl, 2015-07-22:
Ah....
I know where those empty slots came from.
When I created that model I tried to fill it from companion.
AFAIK I did this correct, but when I connected the receiver I got 2 new sensors for Alt & VSpd

I then chose to delete these 2 from the radio.

fraternl, 2015-07-23:
Hi Bertrand,

I was interested in the way it was fixed and wanted to try it first thing in the morning.
But then I discovered that no nightly build was made.
This may be related to the new way of creating nightly builds.
Just speculating...
Maybe you can check it and force a new run?

Could you tell me how it is fixed?
Is Companion changed to handle the gaps in the list or is the firmware changed so no gaps will be created.

I didn't want to manually change the rows of the entries because I read some of the parameters are some of the properties of the sensor are not shown in Companion, but are there....
Is this true or did I misinterpret?

bsongis, 2015-07-23:
I am in the middle of a work on nightly builds, today the firmware part is ready, but still not Companion, you have to compile it yourself, or wait for one or two more days!

fraternl, 2015-07-23:
I think I will wait.  :D 

I will be getting my first 3D-printer and will be busy with that I think.
It might do you good to have a few days off reading my issues

projectkk2glider, 2015-07-23:
@fraternl I hope you are only kidding :smile: 

Your bug reports are invaluable, although it may look like you are bothering the devs, we actually want to have as many bug reports as possible. That is the only way we are able to fix them. Thanks for your efforts (also to all the other testers/bug reporters).

fraternl, 2015-07-24:
The "Print Model" with the "Inconsistent Parameter" is still not fixed in  Companion 2.1.1N390
You could use my EEPROM as a test...

The rest is working now.

bsongis, 2015-07-24:
Did you check the build date?

fraternl, 2015-07-24:
Yes, July 24th....

But the main problem is fixed
Only the bug in "Print Model" is there...  The "nconsistent Parameter"

When I made this ticket they seemed to be related.
It looks now they were not.

kilrah, 2015-07-24:
And what is it? Does it write you "inconsistent parameter"?

fraternl, 2015-07-24:
Yes, the screenshot is given in the first post

kilrah, 2015-07-24:
OK, do you have your eeprom posted somewhere? Can't reproduce with my test cases, I get [Tx] where x is the telemetry slot.

fraternl, 2015-07-25:
[image]

I'm only working with the model "Cumel" in 2.1.x

When this little bug has been sorted I would like to get rid of these empty slots.
I also would like to see how RxBat now works when it autodiscovers it. I believe Bertrand changed the amount of positions to 2 anyway.
Is it the same if I do this by hand in Companion? I believe reading that there are hidden attributes that are not shown in Companion. Is this true?

projectkk2glider, 2015-07-26:
@fraternl try to post only relevant information for this particular issue here. If you discover a new/unrelated problem it is best to create a new issue. It helps in tracking and makes sure that all issues get done. Otherwise the issues might get missed and not fixed.

fraternl, 2015-07-26:
Closed this ticket because main issue was solved.
Opened a new ticket (https://github.com/opentx/opentx/issues/2570) regarding the unsolved part.

commit a100af81 (2015-07-22): Fixes #2541
closed 2015-07-26 by fraternl