opentx/opentx #2542: Logging in OpenTx 2.1

flarssen, 2015-07-22:
When changing which parameters to log, data will not be valid if a logfile for that day is existing, since new data will not match labels in first line. A new file should be created in such cases.

bsongis, 2015-07-23:
Right

projectkk2glider, 2015-07-23:
What if we leave things as they are. The logging settings usually don't change very often anyway. And if user changes them, then it is his responsibility to either delete existing log file for current day or rename it  before enabling telemetry log with the new items.

flarssen, 2015-07-23:
That would cause frustration, as mostly those who already have experienced this would be aware of it. 
Log settings will change whenever new sensors are added, assuming one remember to enable logging for it. 
IMO, logging should default ON for new sensors, as It's better to have too much data by accident, than loosing data.

Bamfax, 2015-07-23:
I would not mind if the logfile is unique for every time I flip the switch, as there are normally new flightsessions. Normally, when having multiple flights, I split the csv in separate logfiles anyway. So I would think that a filename "modelname-date-id++" would be great. But that might be a change people see quite different.

projectkk2glider, 2015-07-27:
@bsongis Sorry, but I can not see how the commit https://github.com/opentx/opentx/commit/98b4d266768ca022ae0b7320391817b1164d8ac8 fixes this issue. 

The logs are only closed, but as soon as new line is about to be written the old log file is still found , the new header is not written and the new line is just appended. Exactly the same behavior as before.

bsongis, 2015-07-27:
The next call to writeLogs() should see the logs file closed and call openLogs. openLogs should write again the headers.

bsongis, 2015-07-27:
Well in fact I am afraid the filename will be the same (same date), f_size will return a positive value and we will have no headers ...

It would mean that one file per day is not so good ...

flarssen, 2015-07-28:
create filename 
while file exists
    create filename(x)  (x = 2 - >)

bsongis, 2015-07-28:
Well. If I am right today when we change the current model and then, within the same day, we return to  it, we just continue writing in the same logfile. Am I right in saying that we should also change in this case? I mean open a new file each time openLogs is called...

flarssen, 2015-07-28:
At every start of logging:
Build the labels.
Find last file that matches [model]-[date]*.csv.
If there is a file, compare it's labels against current labels.
If labels match, start appending data.
If labels doesn't match, or there is no file, create a new file:.[model]-[date](x).csv

This means a new file whenever there has been a change in number of sensors (columns), sensor names or units of measurements.  The last one is perhaps stretching it a bit, and means units has to be added to all labels where this can change, like it's done already for GSpd and Galt.

projectkk2glider, 2015-07-28:
All that software processing and overhead for a feature that will get used very rarely in practice. For a particular model, how many times an average user will change which fields are logged? Maybe a couple of times in the beginning of model set-up and probably never again. 

I had this situation on the field yesterday. I enabled logging of some new telemetry fields. Then after I was done I simply deleted the current log file (via SD card manager while the logging was off) and started logging again.

flarssen, 2015-07-28:
If someone decides to implement this anyway, i suggest to also put a "One logfile pr. session" checkbox somewhere, ref. Bamfax's comment above.

CoyotteDundee, 2015-07-29:
Hi all,
I do not know OpenTx Internals so this proposal may not be applicable but, would'nt be easier to set a bit somewhere in the plane model whenever either one parameter to log is changed.
At the start of  logging, if this bit is set, a new file is created with headers and the bit is cleared.
Otherwise, the process runs as usual.

Coyotte

flarssen, 2015-07-29:
I was thinking the same after my last comment. A new file if something has changed or user want one file pr. session. No need for reading and comparing labels.

commit 98b4d266 (2015-07-23): Fixes #2542
closed 2020-07-07 by stale[bot]