opentx/opentx #2544: Invalid model and settings file  OpenTX Companion 2.1

rguptaca, 2015-07-23:
Tx Taranis X9D
Windows 8 (up to date)
Downloaded OpenTX Companion 2.1
Downloaded opentx-taranis-timer3-noheli-mixersmon-ppmus-en-2.1.0 (using companion)
Turned on radio with THR Trim & ELE Trim pushed to centre first
Plugged in USB cable
Wrote new firmware
Unplugged USB cable
Turned on radio is the normal way
Turned off radio
Turned on radio with THR Trim & ELE Trim pushed to centre first
Plugged in USB cable
Read models and settings from radio
(did nothing)
Wrote models and settings to radio

Errors
Invalid model and settings file (shows a path to a temp directory 3744-compact.bin
Close that window (by pressing ok)
The radio firmware belongs to another product family, check file and preferences.

rguptaca, 2015-07-23:
[image]
[image]
[image]
[image]

bsongis, 2015-07-23:
Do you have a Taranis or Taranis Plus?

rguptaca, 2015-07-23:
Taranis

Rav

Have a wonderful day!

Ravi Gupta, PMP
Founder of Hike For Happiness
A Mental Health Fundraiser
http://hikeforhappiness.ca/
(519) 697-0796

g0ggg, 2015-07-23:
Hi Bertrand
I'm seeing the same issue with new just received X9E, radio profile set to x9e and comp 2,1 on Win7.1.
Tx  - as supplied taranis -x9e-2.1.0 ust upload config and unmodified download with check compatibility =  same results as above.

note companion does not display the Global Functions and the simulator does not have the new sliders.
Hope this is useful, I can run tests if required as my main tx is x9d
Regards neil

bsongis, 2015-07-23:
You are right, I just reproduced it.

g0ggg, 2015-07-23:
For information
I also manually loaded the new FW  from the companion download using MikeB technique from SD card and I still had the error.

many thanks the X9e displays are excellent

bsongis, 2015-07-23:
Did you choose Taranis X9E as Radio Type? The bug I found occurs when you select a wrong radio type, but if I choose Taranis X9E, it's all good.

bsongis, 2015-07-23:
Would you attach a screenshot of the VERSION screen of your X9E? There is a screenshot function in Custom Functions that you can use ...

g0ggg, 2015-07-23:
Hi Bertrand 
X9E is set in the radio profileI've enclosed the complete process and error messages and the sim screen
Thanks neil

g0ggg, 2015-07-23:
email with x9E screenshots sent
version  opentx-taranis-x9e-2.1.0
date 2015-07-23
time 18:31:36
eepr   217
thanks neil

g0ggg, 2015-07-23:
further information, the config download appears to work ok if you do not tick the compatibility box

bsongis, 2015-07-23:
Do you mean the config upload?

g0ggg, 2015-07-23:
'Write Models and Settings to the radio'
Providing I don't check the firmware compatibility box , it downloaded ok
As soon as the box is tick it looks to download but reports the 2 error messages above and the modification is not applied to the tx
As the default is not ticked I guess it has been noticed, its low priority I just want to check for the future when my 9xd goes to 2.1. from 2.0.17
Thanks neil

rguptaca, 2015-07-23:
So you mean when writing from companion to Tx. 

Upload / download is depends on ones point of view:)

Rav

Have a wonderful day!

Ravi Gupta, PMP
Founder of Hike For Happiness
A Mental Health Fundraiser
http://hikeforhappiness.ca/
(519) 697-0796

bsongis, 2015-07-23:
Also the problem occurs only on Windows

g0ggg, 2015-07-23:
yes correct

   

http://hikeforhappiness.ca/

g0ggg, 2015-07-23:
I'm fully patched up windows 7 sp1 
thanks neil

bsongis, 2015-07-23:
@projectkk2glider If you are on Linux right now, would you check with Valgrind?

bsongis, 2015-07-23:
@projectkk2glider  I just found eepromVariant was not initialized, I think the problem is there.

closed 2015-07-23 by bsongis

bsongis, 2015-07-24:
Nightly builds are there, would you check the issue is fixed?
http://downloads-21.open-tx.org/nightly/companion/

g0ggg, 2015-07-24:
Sorry I am not around this weekend but I  will try when I'm back on Tuesday

Thanks Neil 

http://downloads-21.open-tx.org/nightly/companion/

rguptaca, 2015-07-24:
Works
When writing to the radio I checked all boxes. Works.
Minor(?) observation. When installing 2.1 it wants to install to the 2.0 directory.
See image.
[image]

kilrah, 2015-07-24:
What version did you have installed before the upgrade? This could happen if it was earlier than 2.0.17 as that's the one that prepared the terrain for concurrent installs.

rguptaca, 2015-07-25:
As per thread. 2.0

Rav

Have a wonderful day!

Ravi Gupta, PMP
Founder of Hike For Happiness
A Mental Health Fundraiser
http://hikeforhappiness.ca/
(519) 697-0796

kilrah, 2015-07-25:
Need to know which revision of 2.0.

rguptaca, 2015-07-25:
The one from the Website. Ill have to look on my computer... Could take some time for me to get back to you...

Rav

Have a wonderful day!

Ravi Gupta, PMP
Founder of Hike For Happiness
A Mental Health Fundraiser
http://hikeforhappiness.ca/
(519) 697-0796

rguptaca, 2015-07-25:
I believe it may have been pre 2.0.17

projectkk2glider, 2015-07-26:
>When installing 2.1 it wants to install to the 2.0 directory.

I experienced the same issue on my test system when I worked on #2482. It appears that the installer program is somehow finding the currently installed Companion 2.0 and tries to install in the same place. On the other hand the paths in cmake for 2.1 are all different. Maybe it is working on the basis of the package name which is still the same in 2.0 and 2.1 versions.

g0ggg, 2015-07-27:
Hi bertrand
I'm back from a long wet weekend in the UK, I tested the issue on 2.1.1IN394 27/7/15 nightly build, I can confirm the compatibility check now loads the model/settings files.Many thanks neil

   

http://downloads-21.open-tx.org/nightly/companion/—

jgorres, 2015-07-28:
Hi g0gg,

I had the same issue

*Errors
Invalid model and settings file (shows a path to a temp directory XXXX.bin)
The radio firmware belongs to another product family, check file and preferences.*

with my Taranis X9D Plus.

I can confirm that the error does ***not*** occur with the build ***companion-windows-2.1.1N394.exe***.

Using this build it is possible to install the firmware without these error messages.

Hope that helps.

Regards & thank you for this great piece of software,
Jörn.

commit 7d2e48a9 (2015-07-23): Fixes #2544