opentx/opentx #2545: Default resolution of RxBat is 1 decimal

fraternl, 2015-07-23:
Default resolution of RxBat is 1 decimal
Probably not a bug, but intended thought over and intended

In some of my DLG's I fly with 1S and created these logical switches to sound either a "low battery" or a "critical battery"

L1 (RxBt < 3.7 ) AND !L2 Delay (2s)
L2 RxBt < 3.6 Delay (2s)

The step between these 2 is 0.1V which is probably fine.
I initially wanted that 3.6V a bit closer to the 3.7V.
I am aware that the precision of the sensor is not that accurate, but the resolution is higher than 0,1V
Isn't it 13,2/255 = 0,051V ?

The measured sample can be either 3,57, 3.62, 3.67
This means there are 2 steps between them. I would prefer to have it only 1 step.
This does give the impression the resolution is much higher and might encourage some to have them so close they are actually the same.

It's probably just fine, but I would like to know if I can just change the resolution in double precision and set it to 3.63 and 3.70 and will it behave like intend to?

And I would also like to know if the decision to make it 1 precision is taken with the same considerations as I wrote before....

kilrah, 2015-07-23:
The decision for showing 1 decimal is that showing 2 would be misleading, if you see 2 the typical random person will expect to have a 0.01V resolution and not necessarily understand that received data has steps of 0.05V.

fraternl, 2015-07-23:
Yes, corresponds with my remark: "Changing it to 2 decimals does give the impression the resolution is much higher and might encourage some to have them so close they are actually the same. In which case the current 1 position is better."

I think the 2 switches I made for a 1S LiPo work fine...
Would you do it the same?
It's a DLG with an X4R and 4 servo's & variometer...

kilrah, 2015-07-23:
I'd likely put a single alarm at 3.5 and call it a day.

fraternl, 2015-07-23:
Cheers

closed 2015-07-23 by fraternl