opentx/opentx #2547: Opentx 2.1 shows the wrong OpenXsensor telemetry cells data

zyren, 2015-07-23:
I was able to acquire and display three cels, and shows sensor ID 28, but there is no display value, can also be seen three ID, 0300,0301 and 0302.
    I try to modify oXs_config.h in #define SENSOR_ID, 0x1B to 0xBA, just get the ID is displayed as 27, but the problem remains.
    But before Opentx2.0.17 is completely normal, they do not know can not solve this problem in the next version.
[image]
[image]
[image]
[image]

kilrah, 2015-07-23:
What receiver? oXs connected through smart port or hub-style? What oXs firmware version? Can you post your oxs_config.h?

Looks like you might have configured the oXs to act like 3 different multi-cell sensors at the same time instead of just one.

zyren, 2015-07-23:
Sorry, just I did not write a complete and is connected to the S.PORT X8R
, OXs_config.h the settings:

// --------- 9 - Data to transmit ---------
// General set up to define which measurements are transmitted and how
#define SETUP_DATA_TO_SEND \
                         DEFAULTFIELD, CELLS_1_2, 1, 1, 0, \
                         DEFAULTFIELD, CELLS_3_4, 1, 1, 0, \
                         DEFAULTFIELD, CELLS_5_6, 1, 1, 0...

zyren, 2015-07-23:
// Configuration file generated by OpenXsensor Configurator v2.1 the: 23-07-2015
// !! This file is only compatible with version v2.x of OpenXsensor !!

// OpenXsensor https://code.google.com/p/openxsensor/
// started by Rainer Schloßhan

//***********************************************************************************************************************
// Another file in this project (see oXs_config_description.h) provides detailed explanations on how to set up this file.
//***********************************************************************************************************************

#ifndef OXS_CONFIG_h
#define OXS_CONFIG_h

// --------- 1 - FrSky device ID when Sport protocol is used ---------
#define SENSOR_ID    0x1B

// --------- 2 - Serial data pin choice ---------
#define PIN_SERIALTX    4       // The pin which transmits the serial data to the FrSky telemetry enabled receiver

// --------- 3 - PPM settings ---------
//#define PIN_PPM               // Arduino can read a PPM Signal coming from Tx. This allows to change the vario sensitivity using a pot or a switch on TX.
#define PPM_MIN_100     988     // 1500 - 512 ; pulse width (usec) when TX sends a channel = -100
#define PPM_PLUS_100    2012    // 1500 + 512 ; pulse width (usec) when TX sends a channel = +100

// --------- 4 - Vario settings ---------

// ***** 4.1 - Connecting 1 or 2 MS5611 barometric sensor *****
//#define VARIO                 // set as comment if there is no vario
//#define VARIO2                // set as comment if there is no second vario

// ***** 4.2 - Sensitivity predefined by program *****
#define SENSITIVITY_MIN       50
#define SENSITIVITY_MAX       50
#define SENSITIVITY_MIN_AT    20
#define SENSITIVITY_MAX_AT    100

// ***** 4.3 - Sensitivity adjusted from the TX *****
#define SENSITIVITY_MIN_AT_PPM    10   // sensitivity will be changed by OXS only when PPM signal is between the specified range enlarged by -5/+5
#define SENSITIVITY_MAX_AT_PPM    40
#define SENSITIVITY_PPM_MIN       20   // common value for vario is 20
#define SENSITIVITY_PPM_MAX       100  // common value for vario is 100

// ***** 4.4 - Hysteresis parameter *****
#define VARIOHYSTERESIS    5

// ***** 4.5 - Vertical speeds calculations *****
#define VARIO_PRIMARY              0    // 0 means first ms5611, 1 means second ms5611 , 2 means vario based on vario 1 + compensation from airspeed
#define VARIO_SECONDARY            1    // 0 means first ms5611, 1 means second ms5611 , 2 means vario based on vario 1 + compensation from airspeed
#define SWITCH_VARIO_MIN_AT_PPM    10
#define SWITCH_VARIO_MAX_AT_PPM    90

// ***** 4.6 - Analog vertical speed *****
//#define PIN_ANALOG_VSPEED       //  the pin used to write the vertical speed to the Rx A1 or A2 pin (can be 3 or 11 because it has to use timer 2)
#define ANALOG_VSPEED_MIN    -3
#define ANALOG_VSPEED_MAX     3

// --------- 5 - Airspeed settings ---------
//#define AIRSPEED    MS4525

#define AIRSPEED_RESET_AT_PPM      100

#define COMPENSATION_MIN_AT_PPM    60
#define COMPENSATION_MAX_AT_PPM    90
#define COMPENSATION_PPM_MIN       80
#define COMPENSATION_PPM_MAX       120

// --------- 6 - Voltages & Current sensor settings ---------

// ***** 6.1 - Voltage Reference selection (VCC or 1.1V internal) *****
//#define USE_INTERNAL_REFERENCE    // Select the voltage reference, comment the line to activate the VCC voltage reference

// ***** 6.2 - Voltages analog pins *****
#define PIN_VOLTAGE_1    0
#define PIN_VOLTAGE_2    1
#define PIN_VOLTAGE_3    2
#define PIN_VOLTAGE_4    3
#define PIN_VOLTAGE_5    6
#define PIN_VOLTAGE_6    7

// ***** 6.3 - Voltage measurements calibration parameters *****
#define OFFSET_1             0         // offset in mV
#define MVOLT_PER_STEP_1     4.80
#define OFFSET_2             0         // offset in mV
#define MVOLT_PER_STEP_2     9.60
#define OFFSET_3             0         // offset in mV
#define MVOLT_PER_STEP_3     14.40
#define OFFSET_4             0         // offset in mV
#define MVOLT_PER_STEP_4     19.20
#define OFFSET_5             0         // offset in mV
#define MVOLT_PER_STEP_5     24
#define OFFSET_6             0         // offset in mV
#define MVOLT_PER_STEP_6     28.80

// ***** 6.4 - Number of lipo cells to measure (and transmit to Tx) *****
#define NUMBEROFCELLS    6

// ***** 6.5 - Current sensor analog pin *****
//#define PIN_CURRENTSENSOR

// ***** 6.6 - Current sensor calibration parameters *****
#define OFFSET_CURRENT_STEPS         0
#define MAMP_PER_STEP                0.0   // INA282 with 0.1 ohm shunt gives 5000mv/A 

// --------- 7 - RPM (rotations per minute) settings ---------
//#define MEASURE_RPM

// --------- 8 - Persistent memory settings ---------
//#define SAVE_TO_EEPROM
//#define PIN_PUSHBUTTON

// --------- 9 - Data to transmit ---------
// General set up to define which measurements are transmitted and how

#define SETUP_DATA_TO_SEND    \
                        DEFAULTFIELD , CELLS_1_2 , 1 , 1 , 0 , \
                        DEFAULTFIELD , CELLS_3_4 , 1 , 1 , 0 , \
                        DEFAULTFIELD , CELLS_5_6 , 1 , 1 , 0

// --------- 10 - Reserved for developer. DEBUG must be activated here when we want to debug one or several functions in some other files. ---------
//#define DEBUG

#ifdef DEBUG
#include "HardwareSerial.h"
#endif

#endif// End define OXS_CONFIG_h

bsongis, 2015-07-24:
It's a bug in openXsensor. I can have a look if the owner needs. If 3 different IDs are shown, it means that there are 3 values, openXsensor should only use one ID! The sensor ID is good.

closed 2015-07-24 by bsongis

zyren, 2015-07-24:
Well, I try to solve some openXsensor code, please provide the correct data frame format a opentx cells can support, very grateful!

zyren, 2015-07-24:
This is debug data:

Cell calculation for cellId 0 Frist Cell = 1922 Second Cell = 1924 Frsky value = 78478260

Cell calculation for cellId 2 Frist Cell = 1929 Second Cell = 1932 Frsky value = 78C78962

Cell calculation for cellId 4 Frist Cell = 1922 Second Cell = 1919 Frsky value = 77F78264

set new data at 15909 5E 6 7 82 5E 6 17 84 5E 6 27 89 5E 6 37 8C 5E 6 47 82 5E 6 57 7F 5E  

set new data at 16108 5E 6 7 82 5E 6 17 84 5E 6 27 89 5E 6 37 8C 5E 6 47 82 5E 6 57 7F 5E  

set new data at 16309 5E 6 7 82 5E 6 17 84 5E 6 27 89 5E 6 37 8C 5E 6 47 82 5E 6 57 7F 5E  

At 16420 Cnt = 24 mVolt 0 = 3845

At 16422 Cnt = 24 mVolt 1 = 3849

At 16426 Cnt = 24 mVolt 2 = 3854

At 16428 Cnt = 24 mVolt 3 = 3865

At 16431 Cnt = 24 mVolt 4 = 3845

At 16434 Cnt = 24 mVolt 5 = 3839

Cell calculation for cellId 0 Frist Cell = 1922 Second Cell = 1924 Frsky value = 78478260

Cell calculation for cellId 2 Frist Cell = 1927 Second Cell = 1932 Frsky value = 78C78762

Cell calculation for cellId 4 Frist Cell = 1922 Second Cell = 1919 Frsky value = 77F78264

set new data at 16508 5E 6 7 82 5E 6 17 84 5E 6 27 87 5E 6 37 8C 5E 6 47 82 5E 6 57 7F 5E  

set new data at 16709 5E 6 7 82 5E 6 17 84 5E 6 27 87 5E 6 37 8C 5E 6 47 82 5E 6 57 7F 5E  

set new data at 16910 5E 6 7 82 5E 6 17 84 5E 6 27 87 5E 6 37 8C 5E 6 47 82 5E 6 57 7F 5E  

At 16959 Cnt = 24 mVolt 0 = 3845

At 16961 Cnt = 24 mVolt 1 = 3845

At 16964 Cnt = 24 mVolt 2 = 3859

At 16966 Cnt = 24 mVolt 3 = 3865

At 16969 Cnt = 24 mVolt 4 = 3845

At 16972 Cnt = 24 mVolt 5 = 3839