opentx/opentx #2549: Companion Mac version

jeta0357, 2015-07-24:
Hi,
I work on Macintosh, 
please, can you tell me when the Mac companion 2.1.x version will be available ?
thanks a lot
Jean

andyjenkinson, 2015-07-24:
It is available now:
http://www.open-tx.org/2015/07/16/opentx-2.1.0/

Please note there is an issue in 2.1.0, the internal simulator is missing (#2500). This should be fixed in 2.1.1, all being well.

bsongis, 2015-07-24:
@andyjenkinson Would you try to compile the current 'master' branch? Your fix is there, and the simulator should also inside.

andyjenkinson, 2015-07-24:
Doing that right now :)

andyjenkinson, 2015-07-24:
Hmm it still doesn't contain the simulator. What make targets are necessary to include it?
"make companion" includes "simulation", is that sufficient?

I can't build "make simulator", it fails looking for the SDL library, But anyway is that for the standalone simulator?

bsongis, 2015-07-24:
Yes it is

jeta0357, 2015-07-24:
Hi
thanks a lot for your work
just for your info,
i downloaded the Img package and installed it on Yosemite and Maverick 
system

on both system i have the following window when i run the companion V2.1 
program

i click Ignore
and
after that the window is closed and program also

thanks
Jean

andyjenkinson, 2015-07-25:
Can't see an image on my phone but will try to have a look tonight

kilrah, 2015-07-25:
Someone made another issue with a copy/paste, the error is companion tries to load SDL and fails as it isn't present.

andyjenkinson, 2015-07-25:
Ah yes I did update the DMG shortly after initially uploading it to include SDL in it as I noticed it was missing. Try downloading it again.

kilrah, 2015-07-25:
Still same.

kilrah, 2015-07-25:
OK here's the problem:
>Dyld Error Message:
  Library not loaded: @executable_path/../Frameworks/SDL.framework/Versions/A/SDL

You've put SDL right in the Frameworks directory, not in /Frameworks/SDL.framework/Versions/A/ where it's expected.

If I move it to the right location it works fine.

andyjenkinson, 2015-07-25:
Yep just looked and realised the same thing - cp command fail :/

I have uploaded a new version of 2.1.0. For 2.1.1, need to get to the bottom of why simulator is not included.

bsongis, 2015-07-25:
I commited the fix for simulators

kilrah, 2015-07-26:
I've now set up a build environment from scratch on a VM and updated the wiki page with exactly what I've done. 
Companion from current master is now built correctly with frameworks correctly added (runs fine on my non-dev mac). Simulators build and work, the simulator is also built and works but the frameworks are missing in it so it will only work on the dev env. I believe we just need to add the frameworks in the simu, then add it to the dmg and we should be pretty good.

projectkk2glider, 2015-07-26:
I think OSX installation package is still missing simulator libraries. Like it was done in #2500 for Linux, same must be done for OSX.

The crucial changes for Linux were:

The inclusion of simulator libraries in final package in companion/src/firmwares/opentx/simulator/CMakeLists.txt:
```
IF(${CMAKE_SYSTEM_NAME} MATCHES "Linux")
  INSTALL( TARGETS ${OPENTX_LIBRARIES} LIBRARY DESTINATION ${SIMULATOR_LIB_PATH} )
ENDIF(${CMAKE_SYSTEM_NAME} MATCHES "Linux")
```

Then the code that searches for available simulator libs must be modified to search in the right path:
```c
void registerSimulators()
{
    // code snipped

#if !( (!defined __GNUC__) || (defined __CYGWIN__) )
  if (!simulatorsFound) {
    dir = SIMULATOR_LIB_SEARCH_PATH;
    foreach(QString filename, dir.entryList(filters, QDir::Files)) {
      registerSimulator(filename.prepend(dir.path() + "/"));
      simulatorsFound = true;
    }
  }
#endif

}
```

The variable `SIMULATOR_LIB_SEARCH_PATH` gets set by cmake via file ` companion/src/version.h.in `! 

The equivalent must be done on OSX, but I am not familiar enough to tell where the libraries should go. On Linux they are set in main  companion/src/CMakeLists.txt:
```
SET( SIMULATOR_LIB_PATH ${CMAKE_INSTALL_PREFIX}/lib/companion${C9X_VERSION_SHORT} )
```

Which is normally `/usr/local/lib/companion21/`

kilrah, 2015-07-26:
I would think the "usual" way with Mac applications would be to place the simulator libs within the simulator application package.

bsongis, 2015-07-27:
I put the simulators dylib inside the Resources bundle directory, and updated links inside. I think it's still needed to change simulation/simulatorinterface.cpp to search the dynamic libraries inside this directory. Would anybody check this and do the change?

raphaelcoeffic, 2015-07-30:
I'll try to have a look at it tomorrow morning.

PR #2608 (2015-08-01): Fixes #2549 - Use PrefixPath to find the simulator libs
Closes #2549 (*added by @projectkk2glider*)
commit 71530f25 (2015-08-01): Fixes #2549 - Use PrefixPath to find the simulator libs
commit 819509b6 (2015-08-01): Merge pull request #2608 from opentx/raphaelcoeffic/issue2549

Fixes #2549 - Use PrefixPath to find the simulator libs
closed 2015-08-01 by bsongis