opentx/opentx #2552: Companion for Mac - link errors

iflymlnm, 2015-07-24:
Hi, 
I've been trying to build the mac version in my local.
All seems good up until linking, it throws this err.
Any pointers will be appreciated.
-S

Scanning dependencies of target opentx-9xrpro-simulator
[ 98%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-9xrpro-simulator.dir/opentxsimulator.cpp.o
In file included from /Users/sontiveros/src.opentx/opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:239:
/Users/sontiveros/src.opentx/opentx/companion/src/../../radio/src/targets/sky9x/eeprom_driver.cpp:124:20: warning: expression
      result unused; assign into a variable to force a volatile load [-Wunused-volatile-lvalue]
    (void) spiptr->SPI_RDR; // Dump the rx data
           ~~~~~~  ^~~~~~~
/Users/sontiveros/src.opentx/opentx/companion/src/../../radio/src/targets/sky9x/eeprom_driver.cpp:111:18: warning: expression
      result unused; assign into a variable to force a volatile load [-Wunused-volatile-lvalue]
  (void) spiptr->SPI_RDR; // Dump any rx data
         ~~~~~~  ^~~~~~~
In file included from /Users/sontiveros/src.opentx/opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:243:
/Users/sontiveros/src.opentx/opentx/companion/src/../../radio/src/targets/sky9x/coproc_driver.cpp:99:20: warning: expression
      result unused; assign into a variable to force a volatile load [-Wunused-volatile-lvalue]
      (void) TWI0->TWI_RHR ;
             ~~~~  ^~~~~~~
In file included from /Users/sontiveros/src.opentx/opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:258:
/Users/sontiveros/src.opentx/opentx/companion/src/../../radio/src/targets/simu/simpgmspace.cpp:763:35: error: expected
      unqualified-id
        if ((res->d_type == simu::DT_REG) || (res->d_type == simu::DT_LNK)) {
                                  ^
/usr/include/sys/dirent.h:128:18: note: expanded from macro 'DT_REG'
#define DT_REG           8
                         ^
In file included from /Users/sontiveros/src.opentx/opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:258:
/Users/sontiveros/src.opentx/opentx/companion/src/../../radio/src/targets/simu/simpgmspace.cpp:763:68: error: expected
      unqualified-id
        if ((res->d_type == simu::DT_REG) || (res->d_type == simu::DT_LNK)) {
                                                                   ^
/usr/include/sys/dirent.h:129:17: note: expanded from macro 'DT_LNK'
#define DT_LNK          10
                        ^
In file included from /Users/sontiveros/src.opentx/opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:261:
/Users/sontiveros/src.opentx/opentx/companion/src/../../radio/src/telemetry/frsky_d.cpp:416:2: warning:
      "FrSky module alarms removed!" [-W#warnings]
#warning "FrSky module alarms removed!"
 ^
4 warnings and 2 errors generated.
make[2]: *** [firmwares/opentx/simulator/CMakeFiles/opentx-9xrpro-simulator.dir/opentxsimulator.cpp.o] Error 1
make[1]: *** [firmwares/opentx/simulator/CMakeFiles/opentx-9xrpro-simulator.dir/all] Error 2
make: *** [all] Error 2

bsongis, 2015-07-24:
Would you try to change simu::DT_REG and simu::DT_LNK to DT_REG and DT_LNK?

andyjenkinson, 2015-07-24:
See #2271

bsongis, 2015-07-24:
Ok now #2271 is fixed

closed 2015-07-24 by bsongis