opentx/opentx #2553: CYC Mix 2.1.x

marcodcribeiro, 2015-07-24:
Hi,

i upgrade from 2.017 to 2.1.0 but the CYC Mix dont work in Heli CCPM.

bsongis, 2015-07-24:
It should be fixed in latest nightly builds, would you give it a try?
http://downloads-21.open-tx.org/nightly/companion/

griebd, 2015-07-24:
have you set the weight? the heli tab has changed a bit...
in my case I needed to set the input for lateral and longitudinal movement and set the weight, it was default to 0.
from what I understood if you set it to 100 you have the same behavior of 2.0. And if you had any reverse set, then -100.
was that your problem?

bsongis, 2015-07-24:
To download the latest nightly build you follow what I wrote here:
https://github.com/opentx/opentx/issues/2526

marcodcribeiro, 2015-07-24:
The nightly builds works in taranis Radio? if yes i will try later.

bsongis, 2015-07-24:
Yes, you will see how to download the firmware nighly built in my post

marcodcribeiro, 2015-07-24:
Tks,

later i will install companionInstall_2.1.1N390.exe ajnd test

kilrah, 2015-07-24:
Yes the fix was for the swash ring, the rest should normally be ok.

marcodcribeiro, 2015-07-24:
griebd, i did the input for lateral and longitudinal its works. tks

bsongis, tks for the help too

closed 2015-07-25 by kilrah