opentx/opentx #2554: Some errors at Companion 2.11 N390   Profil Taranis Plus   at Ubuntu 13.xx

hrenz, 2015-07-24:
Hy,
Testing now, some errors at Companion: 
- Simulator for this firmware is not available

Some windows are fix, to large, cant reduce it,  
- Konfiguration window
- Heli swashplate mixer window

At Heli mixer is always a fix text  Long cyl and Lat cyl 
but in the language files there are the correct other Terms as Roll, Nick, or Aileron, Elevator 
think there is no translation activated or Log cyl and Lat cyl are in an other File fixed

At ext. Simulator no selection of Transmitter possible only an empty input box

bsongis, 2015-07-24:
Did you compile the simulators?

hrenz, 2015-07-24:
No, not self compiled, ready  *.rpm file downloaded
http://downloads-21.open-tx.org/nightly/companion/

bsongis, 2015-07-24:
No it won't work with RPMs, they are for Fedora!

closed 2015-07-24 by bsongis

projectkk2glider, 2015-07-26:
@hrenz you cannot just unzip the rpm package and start it. The Companion executable now expects and searches for the simulator shared libraries in a specific path. Therefore the way you have done it, it wont find them. 

For a stop-gap solution you could copy all the `libopentx-*-simulator.so` files in the same directory as the `companion21` executable is at. 

Maybe more appropriately you should convert rpm package into debian package with the tool named `alien` https://wiki.debian.org/Alien