opentx/opentx #2555: OpenTX / mac build crashes on startup

dustin, 2015-07-24:
Process:               companion [74725]
Path:                  /Applications/companion.app/Contents/MacOS/companion
Identifier:            companion
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           companion [74725]
User ID:               238135

Date/Time:             2015-07-24 09:04:55.804 -0700
OS Version:            Mac OS X 10.10.4 (14E46)
Report Version:        11
Anonymous UUID:        61497D56-BD3F-F614-1B44-4A3037CE86D2

Sleep/Wake UUID:       41F2B13E-569B-43E9-91EA-535D8B0BC244

Time Awake Since Boot: 71000 seconds
Time Since Wake:       1400 seconds

Crashed Thread:        0

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @executable_path/../Frameworks/SDL.framework/Versions/A/SDL
  Referenced from: /Applications/companion.app/Contents/MacOS/companion
  Reason: image not found

Binary Images:
    0x7fff67fc8000 -     0x7fff67ffe837  dyld (353.2.1) <72A99D0F-0B56-3938-ABC5-67A0F33757C4> /usr/lib/dyld
    0x7fff854bb000 -     0x7fff854bcff3  libSystem.B.dylib (1213) <AD223AEB-237D-35A3-825E-EECF95916838> /usr/lib/libSystem.B.dylib

iflymlnm, 2015-07-24:
Hi, a workaround (that worked for me) is to install SDL (http://www.libsdl.org/download-1.2.php).
Just make sure is the 1.2.x not the 2.x (the later won't work) and Companion boots up.

kilrah, 2015-07-25:
Handled in #2549

closed 2015-07-25 by kilrah