opentx/opentx #2557: Play Value RSSI

MIKE-DAILEY, 2015-07-24:
Not sure if this was an expected change from 2.0.17 to 2.1.0. or not.  When Play Value RSSI in Special Functions is selected with a switch, in 2.0.17 it would play the RSSI value followed with the word "Db".  In 2.1.0 it plays the value only.  I have the July 16, 2015 en American English sound pack installed. Please see attached screen for Ply Value programing. 
[image]

Swwils, 2015-07-25:
This also occurs for any play value - it only says the first value encountered. e.g. a calculated cell value actually being 3.67 volts only says "3 volts".

MIKE-DAILEY, 2015-07-25:
Play Value works correct for RxBt on my system, e.g. plays 5.1 volts.  One of the sound packs fixed that problem for me a few weeks back.

commit 19e68bac (2015-07-25): RSSI lacked unit (Fix #2557)
dBm renamed to dB (made no sense)
closed 2015-07-25 by kilrah